Klipper3d/klipper #3756: NameError: global name 'CommandError' is not defined

jvolkman, 2021-01-11:
I'm debugging a sensorless homing issue, and one problem I keep running into is that `homing.py` seems to be missing some imports.

```
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/safe_z_home.py", line 68, in cmd_G28
    self.prev_G28(g28_gcmd)
  File "/home/pi/klipper/klippy/extras/homing.py", line 179, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/extras/homing.py", line 151, in home_axes
    except CommandError:
NameError: global name 'CommandError' is not defined
Transition to shutdown state: Internal error on command:"G28"
```

Full log: [klippy.log](https://github.com/KevinOConnor/klipper/files/5794291/klippy.log)

KevinOConnor, 2021-01-11:
Thanks.  I think this should be fixed now (commit c57352a3).

-Kevin

jvolkman, 2021-02-02:
Pretty sure this was fixed.

closed 2021-02-02 by jvolkman