Klipper3d/klipper #3759: Resume after pause moves head in XYZ simultaneously instead of first moving Z, and only then XY

jermbe, 2021-01-11:
When resuming a print after pause (filament runout) I noticed that I first need to home all axis before resume.

When I do this, and then perform resume, I notice that the head is moving in all axis. Which almost lead to the printhead bump into my print.

Wouldn't it be better if the printhead first moves to the right Z height and only then in XY?
Even better would of course be that I don't need to home after pause :-)

Hywelmartin, 2021-01-15:
add a longer timeout so that you don't have to home.. And if you have to home just rise the z to higher level..

jermbe, 2021-01-16:
Hey, I managed to find the [idle_timeout] option and have changed it to the following:

```
[idle_timeout]
gcode:
    SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0 # disable the extruder motor
    M104 S0 # set hotend to 0C
#   A list of G-Code commands to execute on an idle timeout. See
#   docs/Command_Templates.md for G-Code format. The default is to run
#   "TURN_OFF_HEATERS" and "M84".
timeout: 600
#   Idle time (in seconds) to wait before running the above G-Code
#   commands. The default is 600 seconds.
```

This seems to have solved my issue. This way I no longer have to home the axis, and my M600 raises Z a bit on pause, so it won't bump into the object anymore :-)

closed 2021-01-18 by klipper-gitissuebot