Klipper3d/klipper #3760: Unable to open config file /home/pi/klipper_config/printer.cfg

krider1, 2021-01-11:
I keep getting this error over and over again,, I edited my config file using putty. I have no experience coding whatsoever so if you can help me like your talking to a 5-year-old
, that would be awesome. Thank you!
[klippy (4).log](https://github.com/KevinOConnor/klipper/files/5797561/klippy.4.log)

Sineos, 2021-01-12:
`IOError: [Errno 2] No such file or directory: '/home/pi/klipper_config/printer.cfg'`

Are you sure that you are having the `printer.cfg` in the path `/home/pi/klipper_config/`. Klipper complains that there is no such file.

krider1, 2021-01-12:
I'm not sure, how do I check?

Sineos, 2021-01-12:
Well, how the hell do you edit it, when you do not know where the file is?

1. Log in via SSH, e.g. with Putty
2. Change directory to the alleged config directory: `cd /home/pi/klipper_config/`
    * If this comes up with an error, then not even the path exists
3. List directory content: `ls` 
    * The file `printer.cfg` should be listed
4. Edit / create the config file: `nano printer.cfg`
4. Please learn at least the basics of Linux. There are many good tutorials out there. 

Side note: 
If you start off using some premade scripts to install moonraker whatsoever be aware that some elements of Klipper's excellent documentation do not work, e.g. the default path to the config typically is `/home/pi/printer.cfg`
In such cases it might be better to revert to the original author of the script for support or follow the official documentation: https://www.klipper3d.org/Installation.html

krider1, 2021-01-12:
So I did the `ls` command and only "client.cfg  client_macros.cfg  moonraker.conf" showed up. I then did `nano printer.cfg` and it showed up blank, whereas when put in that command without `cd /home/pi/klipper_config/` first, I get the printer.cfg file I created before. Is there any way I can just move it to where it's supposed to be?

Sineos, 2021-01-12:
Try `mv ~/printer.cfg ~/klipper_config/`

closed 2021-02-11 by github-actions[bot]