Klipper3d/klipper #3761: Actual probing locations are not taken into account on bed mesh calculations.

arturohernandez10, 2021-01-12:
TLDR; The actual probing locations are not taken into account. Klipper assumes even distribution starting at the corners. My probing points extend to the end on the Y axis but not on the X axis. Causing my first layer to be a mess.

Longer version:

I am pretty sure now I understand what is going on with mesh leveling and it seems like this has affected several other advanced users. BTW, I configured my bltouch myself in Marling and had it working flawlessly for a year. It was the first thing I configured. Since I installed klipper. But I have been having issues since the very first day.

What clued me in was the height on the screen while printing. The z axis variation I saw while moving from one side of the bed to the other through the X axis, did not match the probing done just a minute before. Then I realized this was not on my side. Then I looked at the probing points, calibration output and my config.

My bed is 230x230, and I offset my probing by 30 on the Y and 50 on the X. You can see the probe locations reflect it.  Plus they also reflect the probe offset.

My bed is left intentionally tilted so that on the X axis there is about .9 variation from 50 to 180 on the X axis. My bed protrudes slightly on the center. As I printed a circle on the bed I could see the z axis moving but from far left to far right the height was nearly the same. The width was 100 mm. So if you look at positions 120,110 to 220,110 on the BED_MESH_OUTPUT table they roughly match the 5th row 0.124710 to 0.113086. Which is close to nothing when it should be the two extremes 0.015154 to 0.113086  Which is half my first layer!!!!!!!

Even if my theory was wrong the symptoms are consistent through many many prints. And finally it makes sense that the VORON works since it checks the very edges of the bed.

[bed_mesh]
horizontal_move_z: 5
mesh_min: 50,30
mesh_max: 180,200
probe_count: 4,4
speed: 120

Send: BED_MESH_CALIBRATE
Recv: // probe at 98.000,32.000 is z=0.952500
Recv: // probe at 141.325,32.000 is z=1.037500
Recv: // probe at 184.662,32.000 is z=1.075000
Recv: // probe at 227.987,32.000 is z=1.060000
Recv: // probe at 227.987,88.663 is z=1.095000
Recv: // probe at 184.662,88.663 is z=1.137500
Recv: // probe at 141.325,88.663 is z=1.085000
Recv: // probe at 98.000,88.663 is z=0.995000
Recv: // probe at 98.000,145.325 is z=1.025000
Recv: // probe at 141.325,145.325 is z=1.112500
Recv: // probe at 184.662,145.325 is z=1.145000
Recv: // probe at 227.987,145.325 is z=1.117500
Recv: // probe at 227.987,201.975 is z=1.100000
Recv: // probe at 184.662,201.975 is z=1.155000
Recv: // probe at 141.325,201.975 is z=1.135000
Recv: // probe at 98.000,201.975 is z=1.062500
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
[...]
Send: SAVE_CONFIG
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready
[...]
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // -0.047500 0.037500 0.075000 0.060000
Recv: // -0.005000 0.085000 0.137500 0.095000
Recv: // 0.025000 0.112500 0.145000 0.117500
Recv: // 0.062500 0.135000 0.155000 0.100000
Recv: Mesh X,Y: 10,10
Recv: Search Height: 5
Recv: Mesh Average: 0.09
Recv: Mesh Range: min=-0.0475 max=0.1556
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.062500  0.091111  0.115556  0.135000  0.148611  0.155556  0.155000  0.146111  0.128056  0.100000
Recv:   0.047932  0.080948  0.107324  0.127130  0.140437  0.147317  0.147840  0.142076  0.130098  0.111975
Recv:   0.035679  0.070595  0.098571  0.119815  0.134532  0.142930  0.145216  0.141597  0.132279  0.117469
Recv:   0.025000  0.059969  0.089198  0.112500  0.129691  0.140586  0.145000  0.142747  0.133642  0.117500
Recv:   0.015154  0.048986  0.079100  0.104630  0.124710  0.138476  0.145062  0.143602  0.133232  0.113086
Recv:   0.005401  0.037564  0.068177  0.095648  0.118383  0.134789  0.143272  0.142238  0.130094  0.105247
Recv:   -0.005000  0.025617  0.056327  0.085000  0.109506  0.127716  0.137500  0.136728  0.123272  0.095000
Recv:   -0.016790  0.013064  0.043448  0.072130  0.096874  0.115448  0.125617  0.125149  0.111809  0.083364
Recv:   -0.030710  -0.000180  0.029439  0.056481  0.079282  0.096174  0.105494  0.105575  0.094751  0.071358
Recv:   -0.047500  -0.014198  0.014198  0.037500  0.055525  0.068086  0.075000  0.076080  0.071142  0.060000
Recv: 
Recv: ok
[...]

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  -0.047500, 0.037500, 0.075000, 0.060000
#*# 	  -0.005000, 0.085000, 0.137500, 0.095000
#*# 	  0.025000, 0.112500, 0.145000, 0.117500
#*# 	  0.062500, 0.135000, 0.155000, 0.100000
#*# tension = 0.2
#*# min_x = 50.0
#*# algo = lagrange
#*# y_count = 4
#*# mesh_y_pps = 2
#*# min_y = 30.0
#*# x_count = 4
#*# max_y = 199.97
#*# mesh_x_pps = 2
#*# max_x = 179.99

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5798913/klippy.zip)

[printer.cfg.TXT](https://github.com/KevinOConnor/klipper/files/5798915/printer.cfg.TXT)

toskium, 2021-01-12:
It looks like you are experiencing a similar/same issue as I do and I am totally out of ideas.

I do have a Voron 2.4 working and the bed looks like this screenshot according to my 9x9 bed_mesh:

[image]
If you are interested in the probing output see [1]

I am printing several 30x30x0.2mm squares placed over the bed as follows:

[image]

I am expecting that all printed squares have a proper z-offset. But instead the area 

- No. 1 is a bit too far away from the bed so the extrusions are not touching everywhere
- No. 2 is really far away the extrusions don't really touch each other
- No. 3 is too close to the bed.

My relevant config parts look like this:

```
[probe]
pin: ^z:P0.10
x_offset: 0
y_offset: 25.0
speed: 10.0
samples: 3
samples_result: median
sample_retract_dist: 3.0
samples_tolerance: 0.006
samples_tolerance_retries: 3

[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 25, 25
mesh_max: 275, 275
probe_count: 9,9
mesh_pps: 2,2
algorithm: bicubic
relative_reference_index: 40

```

```
[1] Probe Output
1/12/2021, 12:54:08 PM | probe at 275.000,250.000 is z=0.347500
-- | --
1/12/2021, 12:54:08 PM | Mesh Bed Leveling Complete
1/12/2021, 12:54:08 PM | probe at 275.000,250.000 is z=0.347500
1/12/2021, 12:54:08 PM | Mesh Bed Leveling Complete
1/12/2021, 12:54:07 PM | probe at 275.000,250.000 is z=0.347500
1/12/2021, 12:54:07 PM | probe at 275.000,250.000 is z=0.347500
1/12/2021, 12:54:06 PM | probe at 275.000,250.000 is z=0.347500
1/12/2021, 12:54:06 PM | probe at 275.000,250.000 is z=0.347500
1/12/2021, 12:54:05 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:05 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:04 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:04 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:03 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:03 PM | probe at 243.750,250.000 is z=0.282500
1/12/2021, 12:54:01 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:54:01 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:54:00 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:54:00 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:53:59 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:53:59 PM | probe at 212.500,250.000 is z=0.230000
1/12/2021, 12:53:58 PM | probe at 181.250,250.000 is z=0.270000
1/12/2021, 12:53:58 PM | probe at 181.250,250.000 is z=0.270000
1/12/2021, 12:53:57 PM | probe at 181.250,250.000 is z=0.270000
1/12/2021, 12:53:57 PM | probe at 181.250,250.000 is z=0.270000
[... 441 lines cut ...]
1/12/2021, 12:49:43 PM | probe at 118.750,-0.000 is z=0.340000
1/12/2021, 12:49:42 PM | probe at 118.750,-0.000 is z=0.342500
1/12/2021, 12:49:42 PM | probe at 118.750,-0.000 is z=0.342500
1/12/2021, 12:49:41 PM | probe at 118.750,-0.000 is z=0.342500
1/12/2021, 12:49:41 PM | probe at 118.750,-0.000 is z=0.342500
1/12/2021, 12:49:40 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:40 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:39 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:39 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:38 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:38 PM | probe at 87.500,-0.000 is z=0.300000
1/12/2021, 12:49:36 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:36 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:35 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:35 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:34 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:34 PM | probe at 56.250,-0.000 is z=0.282500
1/12/2021, 12:49:33 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:33 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:32 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:32 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:31 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:31 PM | probe at 25.000,-0.000 is z=0.315000
1/12/2021, 12:49:01 PM | probe at 250.000,25.000 is z=0.289875
```

arturohernandez10, 2021-01-12:
After posting here I attempted to invert the sign of the offset. And extend the size of the bed as much as possible and that helped. But it still is off. I could just straighten my bed. But I want to fix this. I may write guide to visit every probing location and see what the z says on the screen. If I can figure out the math I can fix the code and submit a PR. I also think the documentation and configuration are confusing. And that is causing some of the problems.

arturohernandez10, 2021-01-12:
Btw I think the bicubic implementation is not good. I would stick with the default. The way to know if by sending a BED_MESH_OUTPUT command and looking say the in-between values. In my case they were way off. It may be the tension parameter. Maybe that could be your problem.

arturohernandez10, 2021-01-13:
I created a gcode file to traverse the bed. At a slower speed so that I could see the z value change. And I could easily see the z value change. After multiple tries I finally decided to level my bed as much as possible and disable mesh leveling. The last time I ran my custom gcode. The adjusted z values on the screen were higher on the edges. Add if my bed was concave. While in fact has always been convex like most ender beds. It's possible that configuration could cause this. But there is something wrong going on here.

Arksine, 2021-01-14:
The code has been well tested.  It is certainly possible that there is a bug somewhere, such is the nature of software, but it is unlikely.   The bicubic implementation is cardinal bicubic interpolation.  The amount of curvature interpolated can be increased by increasing the tension option.   At lower tension values it will be closer to bilinear interpolation.  At higher tension values curvature is interpolated using slope, which creates overshoot.  I don't believe that @toskium has an interpolation issue, he has what appears to be probe location bias.  This is common on a Voron if you have twisted X axis.  I would encourage you both to [follow the instructions here](https://www.klipper3d.org/Probe_Calibrate.html#location-bias-check) to check for it.

If you think that the documentation is confusing please feel free to PR a clarification.

As to your issue, your theory is incorrect.  The bed_mesh module makes no assumptions about your bed.  It creates a mesh based on the values entered for `mesh_min` and `mesh_max`.   When klipper is making an adjustment to the z-axis any (X,Y) coordinate outside of of the mesh will be clamped to the nearest value at the boundary of the mesh.

When using the LCD to observe the Z location during a print keep in mind that you are seeing a future location, not the current location of the tool.  This is due to lookahead.   The better way to approach this would be to simply move the tool across the bed manually, then observe the Z position.

> My bed is left intentionally tilted so that on the X axis there is about .9 variation from 50 to 180 on the X axis. My bed protrudes slightly on the center. As I printed a circle on the bed I could [...]

If you look at the probed positions the interpolated values in the mesh are correct.  If you think that probed positions do not match the geometry of your bed then you have an input issue.

arturohernandez10, 2021-02-08:
@Arksine 

Eric,

Thank you so much for your reply. The look ahead explanation would be useful to include in the docs. Even better would be to read the data from further down in the processing pipeline so that it reads the actual value even if it's sampled. It is a bit counter intuitive to see a different value on the screen.

Unless you want to keep this open for the delay. I'm happy closing this issue.

closed 2021-03-08 by arturohernandez10