Klipper3d/klipper #3763: FR - homing_heaters, turn off heater while probing mesh, then heat up before probing again

DroneMang, 2021-01-12:
Hi,

I am getting what I believe is electromagnetic interference from my bed heater while probing with a BLTouch.  The heat mat is a bit smaller than the printable area, and when probing the edges I get different measurements.

In Marlin there is a feature to disable the bed heater while probing, and then heat back up in between probes for a bed mesh, I am wondering if this could be incorporated into Klipper.  I tried incorporating the homing_heaters section, but it doesn't appear that the bed heats back up, I verified this by cancelling the bed besh.  Also, the temps do not update while doing a bed mesh. 
 Thanks!

DroneMang

Log is unrelated.
[klippy7.zip](https://github.com/KevinOConnor/klipper/files/5799405/klippy7.zip)

Sineos, 2021-01-12:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#homing_heaters

DroneMang, 2021-01-13:
Yes, I've seen that link, it works as advertised for homing, but does not heat the bed back up for subsequent bed probes.  The idea is to take current off the bed heater while probing to minimize EMI, but then heat the bed back up for the next probe so that the bed stays at the desired temp.  

homing_heaters just shuts off the heater entirely during a bed mesh calibrate allowing it to cool to room temp, and then heating back up after the bed mesh is complete.  Thanks.

Sineos, 2021-01-13:
I used this option as well (meanwhile deactivated again) and if I remember correctly, it does in fact heat up between the probings. I lost around 5 °C (65 °C -> 60 °C) from beginning to end of the probing. 

My explanation was that apparently the time spent for probing is considerably longer than the time to move from one probing point to the next, so the temperature cannot be stabilized (or only could be stabilized with enormously prolonging the whole process).

DroneMang, 2021-01-13:
Maybe a setting could be added that would force waiting for the bed to heat
back up before probing again rather than just proceeding to the next probe
point and hoping that the bed actually heated up?

Also I am using the BL touch in the touch mode, and the heating up of the
bed could start as soon as the probe triggers not once it has reached the
predetermined height above the bed, this would save a few seconds per probe
and speed up the bed mesh calibration.

I have a rather large bed and because of the EMI I must do 10 probes per
point to get an average.

On my other large printer with Marlin if I utilize this feature that turns
off the heater while probing I got such an accurate first layer that I
could use 0.1 mm first layer height and get a consistent first layer
thickness.

Hywelmartin, 2021-01-15:
is the bed grounded?

DroneMang, 2021-01-15:
It's an aluminum bed, so do you mean run a ground from it to the ground in
the wall? It's a 110V bed. Thanks, this may be a good idea, can you explain
the operating principle of it a bit?

Hywelmartin, 2021-01-15:
If you have 110v alternating current.. you should use PE.. for safety at least.. and for EMI shielding it would be better at least
https://en.wikipedia.org/wiki/Electromagnetic_shielding
https://en.wikipedia.org/wiki/Earthing_system

DroneMang, 2021-01-15:
What do you mean by PE? Thanks.

Sineos, 2021-01-16:
PE = Protective Earth as part of your domestic electrical setup.

@Hywelmartin is refering to:
1. Safety aspect: Prevent dangerous electrical shocks due to fault current in case of a defect (insulation etc.)
2. Means of reducing EMI by allowing induced currents to drain via PE

DroneMang, 2021-01-16:
So if I attach a ground to the aluminum plate that the bed heater mat is attached to, the induced currents in the Aluminum would be grounded and disspate, but wouldn't the bed heater still be producing emi in itself because of the solid state relay SSR switching the bed so fast, I could see this making the situation better, but not removing it.  Thanks.

Sineos, 2021-01-17:
Well, as a private person you have every right to electrocute yourself. IMO it is grossly negligent to have 110V or 220V attached to conductive parts of the printer without proper (0 Ohm) PE connection.

WRT to EMI it will neither prevent nor 100% cancel it but it is a good measure to reduce it.
Also you have to differentiate between the different types / sources of EMI:
- Directly connected: Steppers, PWM etc will pollute your system with EMI directly as they are commonly connected to your entire electronics. This is less the case for a SSR because it decouples your electronics from the bed's current. The SSR is switched (causing EMI) but the currents for switching it are very low.
- Radiation: Every wire with a current will send out EMI as electromagnetic radiation. This can be countered e.g. by twisted pair cables, shielding of cables, proper cable routing etc. In addition proper bonding will also reduce this as the picked up EMI can drain towards ground. The heating mat itself is a pretty good source of EMR as it has kind of an antenna design with high currents that are switched.

DroneMang, 2021-01-17:
I can verify that it is emi from the bed causing the bad bed mesh.  I set the mesh to probe to the same distance from the edges of the heater mat, and my mesh is much better, but now I can't probe the outer 2" of my bed.  Adding this feature to Klipper would greatly improve the meshes.

I have twisted all the wires in the printer, and even found the pairs in the stepper wires and twisted the two loops to reduce emi there.  I've done a good job with grounds, but didn't think to put one from the bed to the PSU, I'll do that, good tip, its not standard in most kits to do this.  I am also going to add a noise suppression filter to the output of the SSR to stop lights flickering in the house, a Crydom 1F25.

DroneMang, 2021-01-17:
I just thought of a possible workaround.

Enable homing_heaters, but set the travel speed to something ridiculously slow, so that the heater has time to heat up in between probes for the mesh during the travel move, and since emi will be reduced because the heater is off, a single probe could be used with the BLTouch, rather than 10 that I have been doing to get an average.

DroneMang, 2021-01-24:
I tried setting the travel speed to 1 and then cancelled BED_MESH_CALIBRATE, with an emergency stop, and the temp had not risen during the travel movement to the temperature set point.  So this is a feature still needed IMO.  Also, the bed temp should be reported during the BED_MESH_CALIBRATE, and you should be able to cancel with an ABORT instead of an emergency stop.  Thanks!

[bed_mesh]
speed: 1

closed 2021-02-23 by github-actions[bot]