Klipper3d/klipper #3765: CoreXY Homing Direction / Origin

welo78, 2021-01-12:
Hello guys, I seem to have a problem wrapping my head around this problem. I cannot seem to get my printer to understand that the current position it is homing to is 0, 210. 

I know that the standard is that the printer is homing to 0,0 but how can I tell it that it should actually home to 0,210 without it swapping its direction ? 

I tried using the position_endstop to be 210 but that just makes it move to the opposite corner as a whole. I would like to either:
1. Home to upper left corner by knowing that these are 0,210 X and Y coordinates
2. Or having it home to the lower left corner to 0,0

Ultimately my goal is to have the lower left corner as origin of 0,0. I cannot seem to wrap my head around this. What should I try?

[image]

[CoreXY config.txt](https://github.com/KevinOConnor/klipper/files/5801322/CoreXY.config.txt)

pierre-vo, 2021-01-13:
Hi @welo78,

position_endstop is the proper configuration for that. I suspect your steppers are not moving in the good direction.
So with position_endstop being 0, 0, it should try to go bottom left and with 0, 210 it should go top left. If this is inverted, the problem is not the endstop setting but the stepper configuration.

kaisertheone, 2021-01-19:
switch cables from left stepper motor to right stepper motor and vice versa.
after that it could be that you have to invertone or both stepper direction but then you should be fine.

closed 2021-01-19 by klipper-gitissuebot