Klipper3d/klipper #3766: StealthChop/SpreadCycle inverted with hardwired SPREAD pin

Loreagle, 2021-01-12:
Hello, I have a problem with my MKS Robin Nano V1.2 and TMC2209 but I think can happen also with other boards and drivers: 

Unfortunately on my board there is SPREAD pin on high level by default so the stealtchop_treshold is inverted as the driver datasheet says:
[image]

I know that this can be solved removing the SPREAD pin on the driver but it can be possible to read the SPREAD_EN state and modify the stealtchop behaviour accordingly? This will semplify the use of the drivers also for those that doesn't know this behaviour.

Loreagle, 2021-01-15:
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/5819482/klippy.log.txt)

mks-viva, 2021-01-20:
@Loreagle Hi! If use uart mode, you need not set it, i think can be change by firmware set. If use standard, you need connect R9(0Ω)
[image]
[image]

Loreagle, 2021-01-20:
@mks-viva Hi and thanks for your answer!

I'm not using MKS drivers and it seems I have the resistance you said soldered so when the SPREAD pin is high I have always stealtchop enabled. 

I confirm to you that I'm using UART on MKS Robin Nano with custom wiring and it works, I just don't want to desolder the resistance or the SPREAD pin so if there is a way trough firmware to set SPREAD to 0 it would be awesome!

[image]

KevinOConnor, 2021-01-20:
My understanding of the spec is that the SPREAD pin only sets the startup settings.  Once klipper explicitly programs the en_SpreadyCycle bit, my understanding is that value is then in use.

You attached a log, but that log doesn't show the tmc2209's programmed in uart mode.

-Kevin

Loreagle, 2021-01-20:
> My understanding of the spec is that the SPREAD pin only sets the startup settings. Once klipper explicitly programs the en_SpreadyCycle bit, my understanding is that value is then in use.
> 
> You attached a log, but that log doesn't show the tmc2209's programmed in uart mode.
> 
> -Kevin

I can assure it works at the opposite if I enable hybrid_treshold... I think is like said in the datasheet, the SPREAD pin set to 1 inverts the behaviour because when I checked the steppers the register was settled with opposite behaviour.
Now to switch off StealthChop I need to have the treshold at zero but I would really like to use the feature without risking to damage my drivers removing the little resistor.

Sorry about the log btw, here a new log: 

[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/5845159/klippy.log.txt)

Thank you!
Lorenzo

KevinOConnor, 2021-01-24:
Okay, wow - that's crazy tmc2209 behaviour.  That might be a question for Trinamic - how can the software reliably disable that behaviour?  Or, have you seen something in the spec that indicates how the software should account for that?

-Kevin

Loreagle, 2021-01-27:
> Okay, wow - that's crazy tmc2209 behaviour.  That might be a question for Trinamic - how can the software reliably disable that behaviour?  Or, have you seen something in the spec that indicates how the software should account for that?
> 
> -Kevin

I just know that when the pin is on high level the behaviour is inverted, so I think that if is possible to read the SPREAD pin status is also possible to invert the treshold behaviour... Of course I am available for testing, let me know if you need anything like registers value, logs ecc.

Lorenzo

closed 2021-02-26 by github-actions[bot]