Klipper3d/klipper #3767: old /dev/gpiomem or /dev/mem support

rbarreiros, 2021-01-12:
Hello,

I have an Odroid C1, and i'm putting it to good use, by installing klipper and octoprint to test how good it can do to an Anet A8 stock.

My problem started when I wanted to implement some failsafes using an output_pin and shutdown_state to turn off the PSU relay. Latest supported kernel for odroid is 3.1.10, I did try to recompile a 5.4.x version but it's a huge hassle to port all changes, I did manage to try someone precompiler 5.3 but /dev/gpiochip? wasn't working at all, so I reverted to 3.1.10 which uses /dev/gpiomem or /dev/mem to access it's GPIO.

I'm ok with porting the current klipper linux mcu code to use gpiomem or mem, but, it would be nice to implement a check when compiling to choose which code to compile, either gpio.c (using /dev/gpiochip?) or other like gpiomem.c (using /dev/gpiomem), asking for some guidance on where to do this.

And finally, it would be amazingly easier to use wiring pi to access the gpio, it makes it easier to install the proper wiringpi for the sbc (odroid version or raspberry pi, etc) and removes all the required checks from the klipper implementation, would this be ok ? or everyone (ultimately Kevin) rather wants it using the low level access for gpio instead ?

Thanks.

KevinOConnor, 2021-01-13:
Maybe @andryblack or @dianlight have some thoughts on this?

-Kevin

rbarreiros, 2021-01-13:
I was sleeping over this last night and thought, how about to make it an option when selecting the linux as mcu in menuconfig, select /dev/gpiochip? subsystem or use libwiringPi ? 

By using libwiringPi it'll check which type of SBC is running and do all it's necessary to access it's GPIO (maybe for I2C and SPI also) and this opens up the possibility to use vendor versions of libwiringPi (like odroid has) and it'll take that responsability off klipper, and keep the /dev/gpiochip? sysbystem intact ?

dianlight, 2021-01-13:
Hi @rbarreiros,
 this is an old subject. You can refer the discussion on the old PR #2787 and #2781.

As you can see the first implementation was based on the old gpio dev access but was abandoned and never merged.  I tried it on an older OrangePi ( first gen for tester with kernel 4.x ) but was very unstable.  I suggest you to try to upgrade the kernel. Refer http://linux-meson.com/doku.php?id=start there is a porting for C1 to kernel 5.x 

L.

rbarreiros, 2021-01-13:
Hey @dianlight,

I'm sorry to bring back an old discussion, I did search for something similar but apparently not well enough....

Did you notice my last comment ? Would an optional alternate version using libwiringPi be feasible ? (I'll implement it ofc)

I ask because I honestly believe It'd be less of a hassle for me to implement it than trying to get that kernel working decently (spent a whole day trying to work out 5.3 and 5.4 kernels mainline and hardkernel patched in C1), and others might benefit from it also.

Thanks.

dianlight, 2021-01-13:
Adding wiringPI as dependencies can be a pain in the ass. As you described there are many version around and also many custom version from the vendors.  
You need not only to implement the gpio access function but also a build-model for the system where you can select wiringpi or gpio and then for wiringpi the version or custom version. 

For example  the wiringPI version for RPi3 and RPI4 differs, there is a custom implementation for Orange Pi H5 and another for H6 etc etc.

It is a lot of work also to maintain, I don't know how many people may be interested in supporting the old model of GPIO.

I suggest you to fork klipper and made your change for C1. You can always use your fork on C1 and align with this repository  if the PR is non accepted.

L.

rbarreiros, 2021-01-13:
Although implementations differ between SBC's the api should remain the same or I guess it would! So the user/system would be responsible to install the correct one for that SBC, klipper should only need to link to the available libwiringPi, am I thinking this wrong?

I'll do as you suggested, thanks.

closed 2021-02-12 by github-actions[bot]