Klipper3d/klipper #3768: Measuring Resonances (feature request sorta)

thelandinlook, 2021-01-13:
Hello @KevinOConnor 

I just purchased an accelerometer in order to measure resonance on my ender 3 v2. A problem that occured to me just in the nick of time is that the test probably was not made for bed slingers (at least for the way I had my accelerometer set up. The problem is that when hot gluing my device to the bed to do the y axis, I was unaware that the print head would move about 20mm above the (almost) middle. This almost crushed my device, which I stopped quickly enough, but it could have been a potential headache. Also I'm aware that an ender 3 probably isn't the target audience for a feature like this, but maybe a warning could be put in the guide, or perhaps even an option to choose how high the z axis sits during this calibration. 

On a less important note, maybe it could be added that a good way to attach the device to the printer is with hot glue, making sure to put electrical tape on all the surfaces you want to glue. (it is surprisingly difficult to get tape off a print bed).

Lastly on a more obscure observation, the Y axis tuning turned off my wifi which is about 2 feet away from my printer, on the same shelf; this happened consistently (about 4 times). Not really a big deal but I thought it was interesting, and perhaps there may be ramifications for other electronics in the area.

Thanks for all your hard work,

one love x

thelandinlook, 2021-01-13:
On second thought, maybe using hot plastic instead of hot glue would reduce vibration reduction to the accelerometer , which is why I imagine a mount is recommended. So maybe some best practices would be useful.

thelandinlook, 2021-01-13:
After securing the chip to the bed with more glue and in such a fashion that should transfer more vibration an inconsistency occurred between that and my initial y test. The first test resulted in the conclusion that I should use 2Hump_EI with 88hz, vibr 4.7%, sm0.12. I repeated this test with SHAPER_CALIBRATE AXIS=X and got the same result.

My second test with TEST_RESONANCES AXIS=Y gave me an almost identical result with 2Hump_Ei 51.4hz, vibr87.7%, sm0.33

However when doing a test immidiately afterwards with SHAPER_CALIBRATE AXIS=y, I was recommended ZV with "Recommended shaper_type_y = zv, shaper_freq_y = 33.8 Hz"
A second shaper_calibration revealed the same ZV result..

Is it likely to get such drastically different results from those two tests?

dmbutyugin, 2021-01-13:
> I just purchased an accelerometer in order to measure resonance on my ender 3 v2. A problem that occured to me just in the nick of time is that the test probably was not made for bed slingers

Why not? It works pretty well on bed slingers, and can improve their performance quite a bit.

> The problem is that when hot gluing my device to the bed to do the y axis, I was unaware that the print head would move about 20mm above the (almost) middle ... but maybe a warning could be put in the guide, or perhaps even an option to choose how high the z axis sits during this calibration.

I am not sure what is being requested here. You actually already specify the exact point to run the measurements at yourself (according to the [guide](https://github.com/KevinOConnor/klipper/blob/master/docs/Measuring_Resonances.md#software-installation)), e.g.

```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None

[resonance_tester]
accel_chip: adxl345
probe_points:
    100,100,20  # an example
```

You are free to change that point however you like.

> On a less important note, maybe it could be added that a good way to attach the device to the printer is with hot glue, making sure to put electrical tape on all the surfaces you want to glue. (it is surprisingly difficult to get hot glue off a print bed).

I think it is typical that the folks put the adxl on some mount. For example, I used the following mount for the bed:
[image]
glued under the bed on the epoxy glue (it is sufficiently temperature resistant).

> This can be useful when printing a mount isn't always so intuitive, such as when the area you'd like to put it has many curves (v2 fan shroud).

It might be possible to install it under the shroud. But I think this is anyway the part that might be very specific to each printer.

> Also It's not obvious that a the third test done overrides the first. I had to do a second y test which seemingly overwrote the initial X test.

I am not sure what you mean by that. But all test files include the timestamp, so the files shouldn't get overwritten.

> My second test with TEST_RESONANCES AXIS=Y gave me an almost identical result with 2Hump_Ei 51.4hz, vibr87.7%, sm0.33

> However when doing a test immidiately afterwards with SHAPER_CALIBRATE AXIS=y, I was recommended ZV with "Recommended shaper_type_y = zv, shaper_freq_y = 33.8 Hz"
> A second shaper_calibration revealed the same ZV result..

You'd need to provide the log like `/tmp/calibration_data_y_20210113_034958.csv` too, otherwise there isn't much to help with. The logs however indicate that SHAPER_CALIBRATE wasn't able to find a decent shaper frequency to reduce vibrations, e.g.
```
Fitted shaper 'zv' frequency = 33.8 Hz (vibrations = 91.6%, smoothing ~= 0.136)
...
Fitted shaper '2hump_ei' frequency = 57.8 Hz (vibrations = 88.7%, smoothing ~= 0.162)
```
So it would either seem that there are some problems with the measurements on Y axis (the data fetched by accelerometer is bogus) because of some issues, either with the accelerometer mounting or the Y axis itself. Or that the resonance frequency is so low that it is sort of 'unsupported' by the shapers. In either case, it could have been the case that one measurement turned 2hump_ei shaper to be slightly better and it was chosen, and next time zv was chosen (because 2hump_ei was insufficiently better). I suggest that you plot the charts of resonances for Y axis and take a look at them.

thelandinlook, 2021-01-13:
It seems I can't read very well. Thank you. I'll do some more tests.

closed 2021-01-13 by thelandinlook