Klipper3d/klipper #3769: MCU 'mcu' shutdown: Timer too close

DroneMang, 2021-01-13:
https://github.com/KevinOConnor/klipper/issues/3316

https://github.com/KevinOConnor/klipper/issues/3331

Having the same issues as these, these are closed so opening a new issue.  This started happening yesterday after changing settings with the bed mesh.  I made the probed area smaller, but klipper should extrapolate out to the edges right?  The attached log shows the problem.  Thanks.
[klippy8.zip](https://github.com/KevinOConnor/klipper/files/5806095/klippy8.zip)

-DroneMang

mngr0, 2021-01-14:
I am having the same issue, but my log looks different
I am new here, so I may have made some naive mistake....

I am on rpi3 and arduino2, with klipper/config/example.cfg with some modification for analog pin and adc range.

I try to write "status" in the octoprint terminal 
`Recv: // Can not update MCU 'mcu' config as it is shutdown`
`Recv: // Once the underlying issue is corrected, use the`
`Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the`
`Recv: // config, and restart the host software.`
`Recv: // Error configuring printer`
`Recv: // Klipper state: Not ready`
`Recv: !! Can not update MCU 'mcu' config as it is shutdown`

here is my conf and my log

[klippy.log](https://github.com/KevinOConnor/klipper/files/5814425/klippy.log)

[printer.cfg](https://github.com/KevinOConnor/klipper/files/5814427/printer.cfg.txt)

DroneMang, 2021-01-16:
I figured out that it was the gcode file I was trying to print that was causing this failure.  Here is the gcode if developers are interested, I don't know why the gcode would cause this.  It was produced by the arc welder plugin.
[FailedPrint.aw.zip](https://github.com/KevinOConnor/klipper/files/5825037/FailedPrint.aw.zip)

DroneMang, 2021-01-18:
Closing because I think this is an Arc Welder issue.

closed 2021-01-18 by DroneMang