Klipper3d/klipper #3770: Prusa default config

pollock1234, 2021-01-13:
Hi there,

I've just installed Klipper on my prusa but seem to be having an issue with the menu filament feed menu item at the bottom. It doesnt seem to like the name parameter. The only way for me to get it working is to remove the item completely. Is there a syntax change needed to be made? The attached logfile directs me to the name section as it expects end of statement but receives a : instead.

Thanks in advance

[log.txt](https://github.com/KevinOConnor/klipper/files/5808554/log.txt)

nunorosado1991, 2021-01-31:
Can you add your ptinter.cfg?
is connect via usb?
witch board?
thansk a lot

pollock1234, 2021-01-31:
Hi sorry for the late reply. I'm away from my machine. Its an einsy rambo clone for a prusa mk3s running the stock printer config supplied by prusa. I'll link to their github. https://raw.githubusercontent.com/PrusaOwners/klipper/master/config/printer-prusa-i3-mk3-2017.cfgCheersP

pollock1234, 2021-01-31:
Sorry its a einsy rambo 1.1b if that makes any odds

pollock1234, 2021-01-31:
Sorry its via usb too

closed 2021-03-01 by github-actions[bot]