Klipper3d/klipper #3771: [FR] Monitor the tacho signal of an 3 or 4 wire fan to make sure the hotend doesn't clog

xkuyax, 2021-01-13:
Hey,

i have heard some reports of people that their fan stopped for various reasons (bad connection, fan failed and co) and the hotend clogged then. With hotends which have a delicate heatbreaks (Dragon, Mosquito and co) that clog is rather hard to remove. I think it might sense to implement an feature which monitors the fan via the output signal of an 3 or 4 wire fan (e.g. Noctuas) and throws an error if the fan doesn't spin to prevent clogs.

I had an look at the documentation and config reference, but i couldnt find that feature.

Hardware wise the only thing required is one extra digital input pin and to count the pulses coming in

arkeet, 2021-02-06:
I'm taking a stab at this. I've added a generic pulse counter to firmware (just polling a GPIO pin and periodically reporting the total edge count to the host), and the host calculates the frequency and fan RPM (the fan's tach output is essentially a 2 ppr encoder) and spit out values to the log. The next step is to hook it up to printer fans and allow the user to define RPM limits, so that we shutdown if the fan speed is outside the given range while it's powered on.

Perhaps it can be configured like this?
```ini
[heater_fan hotend_fan]
pin: ...
tachometer_pin: ^...
min_rpm: 6000
max_rpm: 8000 # do we even need this?
```

Some thoughts:
- We need to allow fan to be out of range for a certain amount of time, to allow time for the fan to spin up, but also to give the rotor lock protection time to restart. Perhaps 10 seconds is a good default (my test fan takes over 5s to restart), but should it be configurable?
- Some fans also have a simple stall sensor output (pulls low while the fan is spinning) - may be worth supporting these too.
- A polling frequency of 1kHz should be enough for fans over 10000 RPM. Is this reasonable?
- Would it be worth exploring using GPIO interrupts for edge detection, instead of just polling? This would require new chip-specific code though - could be a future thing.
- Perhaps we should expose the fan RPM to gcode templates and the web api.

xkuyax, 2021-02-06:
> I'm taking a stab at this. I've added a generic pulse counter to firmware (just polling a GPIO pin and periodically reporting the total edge count to the host), and the host calculates the frequency and fan RPM (the fan's tach output is essentially a 2 ppr encoder) and currently just spits out values to the log. The next step is to hook it up to printer fans and allow the user to define RPM limits, so that we shutdown if the fan speed is outside the given range while it's powered on.
> 
> Perhaps it can be configured like this?
> 
> ```ini
> [heater_fan hotend_fan]
> pin: ...
> tachometer_pin: ^...
> min_rpm: 6000
> max_rpm: 8000 # do we even need this?
> ```
> 
> Some thoughts:
> 
> * We need to allow fan to be out of range for a certain amount of time, to allow time for the fan to spin up, but also to give the rotor lock protection time to restart in case of a momentary stall. Perhaps 10 seconds is a good default (my test fan takes over 5s to restart), but should it be configurable?
> * Some fans also have a simple stall sensor output (pulls low while the fan is spinning) - may be worth supporting these too.
> * A polling frequency of 1kHz should be enough for fans over 10000 RPM. Is this reasonable?
> * Would it be worth exploring using GPIO interrupts for edge detection, instead of just polling? This would require new chip-specific code though - could be a future thing.
> * Perhaps we should expose the fan RPM to gcode templates and the web api.

Thank you! You are awesome :D

max_rpm might be an idea to implement, what if the fan fails and sends too many pulses (but tbh i dont think that is ever the case but you know, murphys law :D ) 
1: sounds reasonable its more about long term failure to cool the heatbreak
2: interesting, learned something today 
3: how much communication overhead does this produce?
4: not my expertise field :D
5: if its not too much work that would be nice, could add this to octoprint and display it in the dashboard so you know the fan is running

arkeet, 2021-02-06:
As the noise above indicates, I have produced some code for now at https://github.com/arkeet/klipper/tree/fan-tachometer with initial support for this. @KevinOConnor would you mind providing some thoughts on it?

> max_rpm might be an idea to implement, what if the fan fails and sends too many pulses (but tbh i dont think that is ever the case but you know, murphys law :D )
I don't think this is a plausible failure mode. For now I'm not adding any `max_rpm` setting; it is easy to add one later, but difficult to remove since that would break backwards compatibility.
> 3: how much communication overhead does this produce?
Communication between the host and mcu happens just once a second. The mcu polls the GPIO pin at 1kHz, which shouldn't produce much load (other things like pwm outputs already work at even faster rates). If it turns out to not work well, we can explore using something like GPIO interrupts in the future.

arkeet, 2021-02-07:
Opened a PR: https://github.com/KevinOConnor/klipper/pull/3897

closed 2021-03-09 by github-actions[bot]