Klipper3d/klipper #3773: Skipping steps on Stepper X - layers shifts

cristianku, 2021-01-14:
Any Idea where should I search for the root cause of Skipping micro steps on Stepper X after the last updates of Klipper ( it started around Christmas 2020 . I do regularly update from GitHub ).

Skipping steps meaning that then I have layers shifted on the printed parts. Normally in the middle of a print it shifts of 1 centimetre to the left and then it continues without skipping anymore. Maybe the acceleration is too high ? or the velocity in comparison to the acceleration ( what is the rule here ? )

cristianku, 2021-01-14:
here the klippy.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5813613/klippy.log)

cristianku, 2021-01-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5813625/klippy.log)

cristianku, 2021-01-14:
I have reverted to the version tagged 0.9.1 . Let's see if it helps

cristianku, 2021-01-14:
I see a lot of these messages in the log:

idle timeout gcode execution
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/idle_timeout.py", line 45, in transition_idle_state
    res = self.gcode.run_script(script)
  File "/home/pi/klipper/klippy/gcode.py", line 200, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 256, in cmd_default
    raise gcmd.error(self.printer.get_state_message()[0])
CommandError: Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

cristianku, 2021-01-14:
I have also noticed that the raspberry pi3 cpu is overloaded with the latest commits (  more or less after Christmas )

what have you implemented new that is taking much more processing power ? I think that this is why I am continuously loosing connection to the printer, see the log

KevinOConnor, 2021-01-14:
Unfortunately, we have no way to assist.  The log indicates the software is running as intended (the logs also indicate the cpu load introduced by the software is low).

The best advice we have on "Lost communication with MCU" errors is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

Note that neither high cpu nor communication errors could cause layer shifts.  However, a hardware issue (such as overheating, or voltage instability) could be the root cause of both disconnects and layer shifts.

-Kevin

cristianku, 2021-01-14:
> Unfortunately, we have no way to assist. The log indicates the software is running as intended (the logs also indicate the cpu load introduced by the software is low).

Kevin, please have a better check at the new code., with the older version 0.9.1 from the releases the MCU disconnection disappeared. I saw that you have been working a lot with the last update on differents scripts, so maybe there is something not completely ok. Again with the old version the error disappeared ( but the layer shift is still there, so these are two different things )

KevinOConnor, 2021-01-14:
There hasn't been any notable changes to the communication code nor to the motion planning code in about a year.  It's always possible there could be a subtle error, but we have no way to diagnose that.  None of my printers are misbehaving and none of the other developers are experiencing problems.

If you can reliably reproduce a regression then it's possible to track down what change introduced it.  It is a tool called `git bisect` - if you're willing to go through that procedure (there are a number of guides online describing it) then it will help identify which change introduced the regression.

FWIW, in the past, every "Lost communication with MCU" error that was tracked down was found to be a hardware issue.  Often wires coming loose and causing very unpredictable results.

-Kevin

LazaroFilm, 2021-01-15:
Hi Kevin I'm having the same issue since Xmas #3608 , I'll try to do the`git bisect` you recommended. My issue started before the introduction of the new `rotation_distance` was introduced.

cristianku, 2021-01-19:
the latest Klipper update resolved the problem "lost communication with MCU"

closed 2021-01-19 by cristianku