Klipper3d/klipper #3774: Unknown config object 'adxl345_x'

brotherdust, 2021-01-14:
Hello!
I bought a couple of ADXL345's to measure resonance in my printer.

Following the documentation here: https://www.klipper3d.org/Measuring_Resonances.html

I bought a RPi 4B so I could run both accelerometers at the same time on it (3B has a limitation that SPI1 cannot use mode 3).

I am able to use `ACCELEROMETER_QUERY` successfully if printer.cfg only has a single `[adxl345]`. 

When I try to enable `[adxl345 adxl345_x]` and `[adxl345 adxl345_y]`, then
```
[resonance_tester]
accel_chip_x: adxl345_x
accel_chip_y: adxl345_y
```
I get the following error:
```
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/resonance_tester.py", line 99, in connect
    for axis, chip_name in self.accel_chip_names]
  File "/home/pi/klipper/klippy/klippy.py", line 95, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
Error: Unknown config object 'adxl345_x'
```

I have verified that each ADXL345 works as expected when they are configured individually.

I have tried both software and hardware SPI bus approaches.

Any thoughts here?

Thank you much in advance!

[klippy.log](https://github.com/KevinOConnor/klipper/files/5817344/klippy.log)

dmbutyugin, 2021-01-15:
Oh well, indeed, the documentation needs to be updated. Actually, the full name must be specified, as
```
[resonance_tester]
accel_chip_x: adxl345 adxl345_x
accel_chip_y: adxl345 adxl345_y
```

Then maybe a better naming scheme for bed slingers could be
```
[adxl345 hotend]
...

[adxl345 bed]
...

[resonance_tester]
accel_chip_x: adxl345 hotend
accel_chip_y: adxl345 bed
...
```

brotherdust, 2021-01-15:
@dmbutyugin ,
THANK YOU! That got me a step closer! Now the question is: what's the correct syntax for `ACCELEROMETER_QUERY` now?

dmbutyugin, 2021-01-15:
The short version: `ACCELEROMETER_QUERY CHIP=adxl345_x`, `ACCELEROMETER_QUERY CHIP=bed`

brotherdust, 2021-01-15:
@dmbutyugin , I was just about to say... nevermind.. I read the source code and the docs. It's right there.
Thanks!

So, should I leave this open so @KevinOConnor sees it? Or just close it and update the documentation with a pull request myself, do you think?

brotherdust, 2021-01-15:
Nevermind. Pull request sent.

closed 2021-01-15 by brotherdust