Klipper3d/klipper #3780: PROBE question

jornada812, 2021-01-15:
Hi. I have a question about the PROBE command. Can I save the value this command returns to a variable? I am using a strain gauge so PROBE is giving me the actual offset of the nozzle tip from the bed. 0 runs away if the belts are slightly stretched, the table temperature is different, etc. At the same time, I would like to automate the process of receiving and processing the offset, and not enter it into the config or set it again manually.

Bob90, 2021-01-15:
I'm new, but could you run Probe_Calibrate, get the value to zero, accept and save the value? https://www.klipper3d.org/Probe_Calibrate.html

jornada812, 2021-01-15:
I'm just trying to avoid this. I would like to see something like this in the startup script:

[gcode_macro start_g]
gcode:
   @ Variable = PROBE
   SET_GCODE_OFFSET X = 0 Y = 0 Z = @ Variable
This is a non-working example to explain the essence

Sineos, 2021-01-15:
See https://github.com/KevinOConnor/klipper/blob/64097cb0d0e0a1e2a73f6846b48bae9b509204f1/config/sample-macros.cfg#L53 for an example how to pass a variable to a macro

jornada812, 2021-01-15:
that's fine, just how to return z after PROBE command?

jornada812, 2021-01-18:
I understand that this functionality is missing? Perhaps it should be added? I will give a script when necessary. I have a closed chamber for high temperature printing. Delta printer. I warm up the camera and the table (lengthening of rods, belts, etc.). I see 0 has shifted, have to do a PROBE, write everything into the config by hand, overload the printer and warm up the table again. It is very uncomfortable!

schnello, 2021-01-19:
Good idea. I need also the Z Result in an macro and atm there is no solution.

kaisertheone, 2021-01-19:
i agree, there is a missing feature here.

Have the exact problem.
I want to home the printer towards zmax because i dont want the bed to colide with the nozzle in case of failure...

and i want to set that zmax with the probe.
but cant access the probe result inside a macro.

Sashok071, 2021-01-19:
It would be great to have this functionality

jornada812, 2021-01-21:
The following additions were made to the /home/pi/klipper/klippy/extras/probe.py sources, marked in bold (I did not change anything, just added a little functionality. The essence of the changes - a variable has been added, in which the Z of the sensor triggering is written and this value can now be returned and used in a script):

class PrinterProbe:
    def init (self, config, mcu_probe):
        self.printer = config.get_printer ()
        self.name = config.get_name ()
        self.mcu_probe = mcu_probe
        self.speed = config.getfloat ('speed', 5.0, above = 0.)
        self.lift_speed = config.getfloat ('lift_speed', self.speed, above = 0.)
        self.x_offset = config.getfloat ('x_offset', 0.)
        self.y_offset = config.getfloat ('y_offset', 0.)
        self.z_offset = config.getfloat ('z_offset')
        self.probe_calibrate_z = 0.
        self.multi_probe_pending = False
        self.last_state = False
        **self.last_z_min = 0.**
...
...
...
 def cmd_PROBE (self, gcmd):
        pos = self.run_probe (gcmd)
        gcmd.respond_info ("Result is z =%. 6f"% (pos [2],))
        **self.last_z_min = pos [2]**
...
...
...
 def get_status (self, eventtime):
        return {
     'last_query': self.last_state,
     **'last_z_min': self.last_z_min**
    }

The full text is here https://yadi.sk/d/JIv5O8RLM4sU8w You can open this file in wincsp on raspberries and just replace everything if you search for these points and insert laziness. After making changes to the file, everything must be rebooted for the changes to apply.
Next, in the starting gcode script in the slicer, add this:
G28
G1 Z50.0 F5000
PROBE
start_g

And in start_g we write this:
G1 Z50 F3000
 SET_GCODE_OFFSET X = 0.0 Y = 0 Z = {printer.probe.last_z_min} MOVE = 1

It is important not to use both PROBE and printer.probe.last_z_min in the same script. It is necessary to smash them. Probe into the slicer, use the value in the script. This is due to the peculiarities of the analysis and execution of the script.
All this will allow you to find the actual 0 when replacing the nozzle or there for other reasons for running away 0 without saving the new value to the config, reboots, etc.

schnello, 2021-01-21:
>a variable has been added, in which the Z of the sensor triggering is written
@jornada812 
Does it add the current value or the calculated value? (Result is z=0.300500)

```
Recv: // probe at 142.500,140.000 is z=0.301750
Recv: // probe at 142.500,140.000 is z=0.300500
Recv: // probe at 142.500,140.000 is z=0.300500
Recv: // probe at 142.500,140.000 is z=0.299875
Recv: // Result is z=0.300500
```

jornada812, 2021-01-21:
Recv: // Result is z=0.300500 This

schnello, 2021-01-21:
@jornada812 

Homing doenst work with your probe.py file. I guess this is regarding the latest changes regarding homing in this file
Recv: !! Internal error on command:"G28"

https://github.com/KevinOConnor/klipper/commit/8260a3a0c38e25168e3dfc414ed62fd23e380b21#diff-335c6462be31317d2a13490a9c19bdbab59bc602e3960bd3fd2e9ee9bc0f004e

jornada812, 2021-01-21:
I guess I have an old file. This is what version control and merging is for. But, sorry, I do not know how to use them) Nevertheless, the essence of the additions does not change and they can be done manually

> @jornada812
> 
> Homing doenst work with your probe.py file. I guess this is regarding the latest changes regarding homing in this file
> Recv: !! Internal error on command:"G28"
> 
> [8260a3a#diff-335c6462be31317d2a13490a9c19bdbab59bc602e3960bd3fd2e9ee9bc0f004e](https://github.com/KevinOConnor/klipper/commit/8260a3a0c38e25168e3dfc414ed62fd23e380b21#diff-335c6462be31317d2a13490a9c19bdbab59bc602e3960bd3fd2e9ee9bc0f004e)

schnello, 2021-01-21:
I merge the code and it works ;) (state is my gcode macro)

`
Send: state
Recv: // 0
Recv: ok
[...]
Send: G28
Recv: // action:fan_on
Recv: // action:fan_off
Recv: // probe: open
Recv: // probe at 142.500,140.000 is z=0.297375
Recv: // probe at 142.500,140.000 is z=0.296750
Recv: // probe at 142.500,140.000 is z=0.296125
Recv: // probe at 142.500,140.000 is z=0.296125
Recv: // Result is z=0.296438
Recv: // action:fan_on
Recv: ok
[...]
Send: state
Recv: // 0.296437500001
`

jornada812, 2021-01-21:
Ok I'm glad)

schnello, 2021-01-21:
So can you create a pull request?

jornada812, 2021-01-21:
I can't do that. If you kindly agree to request a transfer, I will be grateful

schnello, 2021-01-21:
Sure but if @KevinOConnor  has questions regarding the code i cant answer it. So please support me in this case

jornada812, 2021-01-21:
Of course, thanks

schnello, 2021-01-21:
But i guess the commit would not merged by Kevin because:
>It is important not to use both PROBE and printer.probe.last_z_min in the same script.

It also the question: Where and how should this be documented.

jornada812, 2021-01-21:
I see this is already documented:

schnello, 2021-01-21:
Ok Please take a look at:
7be6d3ee80250f44e91254d7810dfd172a2edf67
and
9f4500189012e2f44dce188296ac9fec319ca01a

jornada812, 2021-01-21:
> Ok Please take a look at:
> [7be6d3e](https://github.com/KevinOConnor/klipper/commit/7be6d3ee80250f44e91254d7810dfd172a2edf67)
> and
> [9f45001](https://github.com/KevinOConnor/klipper/commit/9f4500189012e2f44dce188296ac9fec319ca01a)

Not sure, but in Python it's important to stick with text formatting. New strings not aligned? I will follow the development of events)

schnello, 2021-01-21:
Hmm it looks for me aligned:
[image]

jornada812, 2021-01-21:
it looks like this https://yadi.sk/i/WBXVVXPU5Gzvxw

schnello, 2021-01-21:
Ah ok.. this is not a tab.. this are spaces.
Please see: https://github.com/KevinOConnor/klipper/compare/master...schnello:master

jornada812, 2021-01-21:
I don't know Python syntax very well. self.last_z_min = 0 or 0.? We need float value. I used "= 0."

jornada812, 2021-01-21:
It is also possible as ".."triggered" during the last PROBE command.." (not QUERY_PROBE)

jornada812, 2021-01-21:
not sure if a space is allowed between pos and [2]

schnello, 2021-01-21:
Code updated:
https://github.com/KevinOConnor/klipper/compare/master...schnello:master

schnello, 2021-01-22:
@jornada812 
Wouldn't as var name "printer.probe.last_z_result" not be more correct?

jornada812, 2021-01-22:
call it whatever you want. I'm more interested in @KevinOConnor's answer)) When does it usually react to suggested changes?

schnello, 2021-01-22:
I dont know. I would suggest we wait some days and preare the pull request. If no feedback arrives, I would release it and discuss it with Kevin in the pull request.

schnello, 2021-01-24:
@jornada812 
https://github.com/KevinOConnor/klipper/pull/3835
Pull request added. Please provide feedback if needed.

jornada812, 2021-02-02:
> @jornada812
> #3835
> Pull request added. Please provide feedback if needed.
How is it going? Do you have any news? What does "All checks have failed" mean?

schnello, 2021-02-02:
Check failed is because a line is longer as 80 chars. Only a minor "issue".
So far no response from Kevin. But some things just take time.

schnello, 2021-02-19:
@jornada812
#3835 is merged

closed 2021-02-24 by jornada812