Klipper3d/klipper #3782: Error on 'SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=': unable to parse

DerInternets, 2021-01-15:
About 80 mins into the print I get an Error Message "Error on 'SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=': unable to parse"

It seems that klipper tries to set the temp on the heatbed to <null> ?
Using Mainsail as the Frontend. Any Ideas?

[klippy(2).log](https://github.com/KevinOConnor/klipper/files/5823163/klippy.2.log)

jakep82, 2021-01-15:
Have you searched the gcode file to see if this command is in there somewhere?

Arksine, 2021-01-17:
If this happens consistently in the same spot then the command is likely in the gcode.  The other option is that Mainsail perhaps erroneously sent this command.  If you have moonraker.log from the time of the print I could look at it and tell you if the command originated from Mainsail.

paulhew69, 2021-01-18:
Just updated MoonRaker this morning and am getting this issue also, all of a sudden.
Both logfiles attached to help.
[moonraker PaulHew 18-1-21.log](https://github.com/KevinOConnor/klipper/files/5829907/moonraker.PaulHew.18-1-21.log)
[klippy PaulHew 18-1-21.log](https://github.com/KevinOConnor/klipper/files/5829908/klippy.PaulHew.18-1-21.log)

paulhew69, 2021-01-18:
[image]

Arksine, 2021-01-18:
This is an issue with Mainsail.   This may be related #3791 as well.

meteyou, 2021-01-18:
@Arksine no, that was not the problem. this issue is, if you delete the number of a heater and press enter without an input. i fixed it already and should fixed in the next release -> https://github.com/meteyou/mainsail/commit/397d9b8a7c4e731eff2610412d8f0239a356c6e8

DerInternets, 2021-01-21:
> @Arksine no, that was not the problem. this issue is, if you delete the number of a heater and press enter without an input. i fixed it already and should fixed in the next release -> [meteyou/mainsail@397d9b8](https://github.com/meteyou/mainsail/commit/397d9b8a7c4e731eff2610412d8f0239a356c6e8)
> 

Thank you, that was most likely the problem.

closed 2021-01-21 by DerInternets