Klipper3d/klipper #3785: Resonance Testing Fails

brotherdust, 2021-01-17:
Been banging my head on this for a few days now and I'm out of ideas.

Basically, when I try to run a resonance test, Klipper reports `Lost communication with MCU` somewhere between 29Hz and 59Hz.

Watching the logs, I can see that `bytes_retransmit` and `bytes_invalid` seem to increment more in proportion to frequency. More frequency = more errors.

Thinking it was a problem maybe with USB, I've tried several different USB cables, different ports. No change.

Tried adjusting `run_current` downwards on the stepper that's performing the test. No change.

If I turn down `max_accel` from `7000` to, say, `5000`... it seems to go a little longer.

`max_accel_to_decel` parameter is set to `7000`, tried `3000`, but Klipper seems to ignore it and set `9000` (see logs).

Could it just be my MCU can't handle the sheer number of commands?

See attached `klippy.log`

Thanks in advance. I really do appreciate all the work that goes in to this project.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5825330/klippy.log)

brotherdust, 2021-01-18:
It appears to be a problem with the EINSY USB interface. I switched to UART1 through a fully-isolated and separate USB to serial adapter. Now works perfectly!

closed 2021-01-18 by brotherdust

KevinOConnor, 2021-01-20:
FYI, the prusa version of the Einsy has a broken usb-to-serial firmware.  Details at: https://github.com/KevinOConnor/klipper/issues/3635#issuecomment-746637372

-Kevin

brotherdust, 2021-01-20:
@KevinOConnor , thanks for that reference! I was looking everywhere for it!