Klipper3d/klipper #3787: Timer too close at startup (power on)

ronald5555, 2021-01-17:
Sorry to disturb you but since a couple of days I have the « timer too close » problem at startup (power on).

Description of my hardware:
1 raspberry Pi 4b 
1 FLYF407ZG driving a
Voron 2.4 350mm printer

Description of the problem
At startup (power on) and only then, I get the message timer too close.
I have to click « restart firmware » in Mainsail and all is OK, Klipper enters the ready state and works like a charm.

What I have done
- created a complete new SD card with
- fresh raspberry lite firmware
- a new Klipper
- a new monraker
- a new mainsail.
- a new USB cable
- controlled the power supplies (24V and 5V have 24.05 and 5.08)
- 
Problem is the same

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5825726/klippy.log.zip)
[moonraker.log.zip](https://github.com/KevinOConnor/klipper/files/5825727/moonraker.log.zip)
[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/5825728/printer.cfg.zip)

KevinOConnor, 2021-01-24:
What changes have you made to the code?  The log indicates something tried to change the beeper pin (EXP1_1 / PB10) before Klipper was ready.  It's not clear to me what attempted to make that change.

-Kevin

ronald5555, 2021-01-24:
I am sorry but I did not change anything to the code.
If I look at the log, I see that PB10 is used by i2c and spi but nowhere I use these interfaces.

KevinOConnor, 2021-01-24:
It looks like I misinterpreted the log of data.  The problem was a bug in rpi_temperature that made temperature_fan produce bad commands.  I'm guessing your local code modification was to enable rpi_temperature with temperature_fan - which then exposed the root bug.

Hopefully, it's fixed now (commit 8de86ad5).

-Kevin

ronald5555, 2021-01-25:
Hello @KevinOConnor 

I was tired of being unable to start my printer without always connect mainsail and send a firmware restart because to the "timer too close" issue.
So I decided to find out what in my configuration is "wrong" or what makes the firmware to be unable to start.
If you have a couple of minutes you can read the following lines else skip directly to the Phase 3 where I think I found the problem/solution.

**Phase 1**
With my actual configuration I get the "timer too close" error.
What is "special" ?
I declared a _[temperature_fan]_ with should be regulated with the RPi temperature. This fan is blowing directly to the RPi because I have an expansion board over the Pi and there is not enough room for convected ventilation.
The RPi temperature is declared directly in the fan section
With this config firmware does not start at power up but starts after sending a firmware_restart command.
The fan works, the RPi temperature is correct.

**Phase 2**
I removed the declaration of the RPi out of the fan section and declared the RPi temperature as a 
_[temperature_sensor RPi]
sensor_type = rpi_temperature
gcode_id = RPi_
in the fan section I leaved _sensor_type = rpi_temperature_
As usual the firmware didn't start due to the "timer too close" error. But as in Phase 1 after a restart everything works.

**Phase 3**
I changed the sensor type in the fan section to _sensor_type = ATC Semitec 104GT-2_ which is a sensor I have free in my cabinet. I also had to ad the _sensor_pin_ of course
And here BINGO, the firmware starts at power up (I tried 10 times). And RPi temperature declared separately as in described in phase 2 is still displayed in Mainsail.

**My conclusion**
Declaring _sensor_type = rpi_temperature_ in a _[temperature_fan]_ does not work. Simply changing the sensor type to a conventional NTC is sufficient.
So I decided to glue my NTC sensor to the top of the RPi processor and now everything is OK.

Attached you will find 3 log files corresponding to each phase

Greetings
Ronald
[klippy-2.log](https://github.com/KevinOConnor/klipper/files/5864941/klippy-2.log)
[klippy-3.log](https://github.com/KevinOConnor/klipper/files/5864942/klippy-3.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5864943/klippy.log)

ronald5555, 2021-01-25:
Hello @KevinOConnor 

As always too fast in writing the above comment.

After doing all that work, I updated my Klipper config and tried to revert to Phase 1.
Now with the new rpi_temperature.py, the firmware starts at power on with the RPi declared as temperature source in a temperature fan.

So my above comment is of no interest.

closed 2021-01-25 by ronald5555