Klipper3d/klipper #3788: MKS Robin Nano 2 + TFT35 v2 show "Booting..."

anteyru, 2021-01-17:
MKS Robin Nano 2 on its screen, when turned on, displays "Booting..."
What type of display should I choose to display useful information for printing?

Hywelmartin, 2021-01-17:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#display-support

Loreagle, 2021-01-17:
> MKS Robin Nano 2 on its screen, when turned on, displays "Booting..."
> What type of display should I choose to display useful information for printing?

I have a Robin nano 1.2 and as I know the display is not supported.

anteyru, 2021-01-17:
> > MKS Robin Nano 2 on its screen, when turned on, displays "Booting..."
> > What type of display should I choose to display useful information for printing?
> 
> I have a Robin nano 1.2 and as I know the display is not supported.

[image]

anteyru, 2021-01-18:
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/5828211/klippy.2.log)

Loreagle, 2021-01-18:
> > > MKS Robin Nano 2 on its screen, when turned on, displays "Booting..."
> > > What type of display should I choose to display useful information for printing?
> > 
> > 
> > I have a Robin nano 1.2 and as I know the display is not supported.
> 
> [image]

Yes it's normal, as I said is not supported yet... if you want to turn the screen off just add this yo your printer.cfg:

[static_digital_output display_reset]
pins: !PC6, !PD13

mks-viva, 2021-01-20:
@anteyru Hi! If use MKS Robin Nano V2 motherboard with klipper, so, you can use LCD12864, because MKS Robin Nano V2 EXP1/EXP2 is supported LCD12864/LCD2004

anteyru, 2021-01-21:
Thanks, all worked well.

anteyru, 2021-01-23:
> @anteyru Hi! If use MKS Robin Nano V2 motherboard with klipper, so, you can use LCD12864, because MKS Robin Nano V2 EXP1/EXP2 is supported LCD12864/LCD2004

Contact designation is different for TFT35 and LCD12864.

closed 2021-02-22 by github-actions[bot]

alexeyzel, 2021-02-27:
@anteyru did you managed to connect TFT35 v2? Can you share your "display" section? I have TS35 v2 screen and can't get it to work.

anteyru, 2021-02-28:
> @anteyru did you managed to connect TFT35 v2? Can you share your "display" section? I have TS35 v2 screen and can't get it to work.
[display_status]

[static_digital_output my_output_pins]
pins: !PC6, !PD13

themacboy, 2021-09-18:
@alexeyzel your TS35 v2 working on klipper?

anteyru, 2021-09-18:
> @alexeyzel your TS35 v2 working on klipper?

Klipper does not support TS35. :(

themacboy, 2021-09-18:
Ok, then we will wait for a solution from @mks-viva hehehe ;)

Spede90, 2022-01-22:
@mks-viva No solution yet?