Klipper3d/klipper #3791: Heated Bed shuts off for no reason.

paulhew69, 2021-01-17:
[klippy17-1-21.log](https://github.com/KevinOConnor/klipper/files/5826201/klippy17-1-21.log)

Hi,

Been using my printer today.
Preparing for a 2nd print, Set heated bed to 100 to warm up.
Came back 10 mins later only to find the heater was off. 
I had not instructed it to turn off.
this is not the first time it has happened, It happens on my RailCore running Klipper on Duet/Duex boards also.

my timeout is 1800, but cannot see how that is relevant as I prewarmed everything this morning for 20mins without it shutting off.

Advice would be appreciated please.

Regards,
Paul

Arksine, 2021-01-17:
I would make sure that the Moonraker client is not directing the heater to cut off.  If you have moonraker.log at the time this occurred I can check to see if it was commanded to shut off.

paulhew69, 2021-01-20:
Morning, and it has just done it again.
Printed, took plate off, removed parts, plate back on, set bed to 100, started heating. Made a coffee, came back and it was off again.
Logs attached to assist!
TIA.
[moonraker.log](https://github.com/KevinOConnor/klipper/files/5841638/moonraker.log)
[klippy(1).log](https://github.com/KevinOConnor/klipper/files/5841639/klippy.1.log)

paulhew69, 2021-01-20:
From the console
[image]

Arksine, 2021-01-20:
Update Mainsail.  If the issue persists please report it on the Mainsail repo, as there is nothing that can be done in Klipper to change this behavior.

closed 2021-02-18 by github-actions[bot]