Klipper3d/klipper #3793: Input Shaper for Long Moves, belt resonance

DroneMang, 2021-01-17:
Hi,

Is it possible to somehow input a signal that will cancel out the resonance on long printing moves at full speed?

I have an FT-6 with very long belts.  At faster printing speeds I get resonances in the belts, I believe.  I can print at 100 mm/s just fine, but at 110 and 120mm/s the resonance gets progressively worse only in certain regions of the bed, and this shows up in the prints.  I have upgraded to 0.9 deg steppers, I have an SKR Pro with 0ms step pulse and actual 256 microstepping without interpolation to TMC 5160 drivers, so there is not much that can be done for the drivers or mcu for smoothness.  I have also added cross braces to the printer frame to help stiffness in the X axis.  I have tuned the input shaper and enabled it, that does help print quality, I had very little ringing prior to enabling input shaper.

I have the belts pretty tight already, but I suspect that tighter belts could help?  I have also thought of designing the carriage to use 2 belts for the X axis one on top of the other to add more stiffness.  Linengineering has some high end Z417 steppers with little cogging torque, these could help put are cost prohibitive.  I believe this will become more of an issue for the large format printers now being created.

Thanks!

Log is unrelated.
[klippy8.zip](https://github.com/KevinOConnor/klipper/files/5826749/klippy8.zip)

brotherdust, 2021-01-18:
@DroneMang , speaking as another Klipper input shaper user, I'm not sure the input shaper would actually be useful for belt resonance. IMO, it's use case is more on the side of "presumably the parts that could cause resonance are constrained as much as possible and there's no other options".

Assuming by FT6 you mean the Folger Tech FT-6, I looked at the specifications. To me, it looks like the belts are too narrow for the design. They specify GT2, which, IIRC is available in 3mm and 6mm widths. Either way, the distances involved (I can't find a specific distance because the manufacturer's site is down, so lets assume the longest axis is, say 1 meter) in this large-format printer would bring the tension high enough on these belts to turn them to bass strings. I'd suggest moving to a 12mm belt, say a Gates PowerGrip GT3. Alternatively, you could put an idler in the middle of the longer belt spans.

Anyway, just my $0.02.

dmbutyugin, 2021-01-24:
Yes, input shaper, unfortunately, cannot compensate for that kind of resonances, because they are not caused by the commanded movement itself. When the toolhead moves at a constant speed, that motion itself cannot cause the vibrations. Belt resonances are likely caused by pulley teeth getting on the belts or by motors vibrations.

Now there could be some different techniques that help avoiding those resonances. Those include motion at varying speeds (to not let the resonance "build-up"), or avoiding some speeds that are more prone to resonances (which may also help with stepper resonances). Unfortunately, varying speed is not practical for 3d printing, because this causes acceleration and deceleration. There are already issues with corners that are compensated (to a degree) with a Pressure Advance, so I guess it is not a good idea to add more potential printing defects to prints by varying the speed. And avoiding speeds that provoke resonances in belts and steppers is, in general, an interesting idea, however its implementation will be quite difficult and likely require reworking of the whole motion planning stack. It is possible, just a lot of work.

bung69, 2021-01-26:
While I expect that some improvement is possible in firmware by slightly varying speed etc, have you considered adding some mechanical dampening to the belt its self?  -  something along the line of a few  soft / children's tooth brushes lightly touching the smooth side of the belt or possibly felt pads.

closed 2021-02-25 by github-actions[bot]