Klipper3d/klipper #3796: TMC 2660 work uncorrectly

kaktus770, 2021-01-18:
Hello! Installed Klipper on clone Duet 2 WiFi, and couldn't tune tmc 2660 on correct work. Initially, after the basic configuration, stepper motors skipped steps while standing still.  This was solved by commenting "idle_current_percent: 30".  Now the stepper motors move, but the microstep parameter in the config affects the amount of movement - at 32 microsteps 12.5mm;  at 64 microsteps 25mm and at 256 microsteps 100mm (set 100mm). Steppers on 0.9 degrees and 20 teeth pulley. As I understand it, in newer versions of the Klipper, changing the microsteps should not affect the amount of movement. 
Dump of drivers with different configurations (DUMP_TMC STEPPER=stepper_a): 
[Dump.txt](https://github.com/KevinOConnor/klipper/files/5828833/Dump.txt)
 Klippy.log with config 16 microsteps.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5828817/klippy.9.log)

Roycinger, 2021-01-21:
Afaik, changing microsteps without any effect on the driven distance will need "interpolate: true"

kaktus770, 2021-01-22:
I tryed to use interpolation (2660 support interpolation only on 16 microsteps), but haven't good result.

fuma55, 2021-01-25:
Problem is still here, changing microstepping effect distance driven by extruder

kaktus770, 2021-01-26:
> Problem is still here, changing microstepping effect distance driven by stepper motor .
> 
> For Example:
> 
> microstepping: 256
> rotation distance: 40 (20 tooth, 2mm)
> full_steps_per_rotation : 400 (0.9 Angle)
> 
> --> printer moves as expected
> 
> microstepping: 16 (dont matter if interpolation is on or off)
> rotation distance: 40 (20 tooth, 2mm)
> full_steps_per_rotation : 400 (0.9 Angle)
> 
> --> Printer moves really slow and only 1/4 of the desired distance.

The problem is that with the necessary microstepping settings, the printer (I have a delta) works at a speed of no more than 50 mm / s, if higher an "MSU error" appears. I partially solved the problem - in low level settings menuconfig set "Step pulse duration 0". The printer work at 256 microsteps, the speed out 300 mm / s. But there was an unpleasant sound from stepper motors - possibly due to the StealthChop mode. As a result, I did what I wanted on the Klipper (extended auto-calibration) and returned to the RRF.

closed 2021-02-25 by github-actions[bot]