Klipper3d/klipper #3798: Display question

ronald5555, 2021-01-18:
Hello, I would like to display the temperature of a sensor and add the 'degree' symbol. 
This symbol is a glyph named  ~degrees~ . 
ow can I add this in the following line: 

name: {"FLYF407Z:%4.0f" % (printer['temperature_sensor FLYF407ZG'].temperature)}

I have tried several methods but I get a nasty horizontal scrolling of the line because the display analyser thinks that the symbol is 9 characters (e.g. the ~degrees~) long instead of only 1 but it displays the ° symbol at the correct place.

I would like to see
FLYF407Z:  25°

attached log file is of no interest for this issue !

[klippy.log](https://github.com/KevinOConnor/klipper/files/5830081/klippy.log)

KevinOConnor, 2021-01-18:
You'll need to attach a log from the event.  If you got it to display, but with weird scrolling, then you'll need to attach the log with that config.

-Kevin

mcmatrix, 2021-01-19:
@ronald5555 
Are you talking about menu item
where you added a name like 
`name: {"FLYF407Z:%4.0f~degrees~" % (printer['temperature_sensor FLYF407ZG'].temperature)}`

ronald5555, 2021-01-20:
Hello McMatrix.
You analyzed und understood 100% correctly the problem.
This is not an important issue, but for the nice looking of the menues it would be great to be able to display that little "°" symbol.

Hope you will find something to do !

Greetings from Switzerland (with a lot of snow)

Ronald

ronald5555, 2021-01-21:
Hello @mcmatrix  
I found a solution to the display of the ° symbol.

After examining the character font, I saw that the symbol is at 0f1h. I tried to include this with a standard escape sequence in my display line but unfortunately somebody strips all the characters with bit 7 set !!
__name: {"FLYF407Z:%4.0f\xf8" % (printer['temperature_sensor FLYF407ZG'].temperature)}_

So I decided to use another character and I choose the vertical bar at 0x7c

[image]

After that I had just to change the character font that Klipper uploads to the display to have the ° symbol located at the place of the vertical bar
_name: {"FLYF407Z:%4.0f\x7c" % (printer['temperature_sensor FLYF407ZG'].temperature)}_

Here the character font see at line 144 I commented out the original vertical bar and inserted the ° bitmap
[font8x14.py.txt](https://github.com/KevinOConnor/klipper/files/5849219/font8x14.py.txt)

With this little hack I have solved my simple problem of displaying the ° symbol in the menu pages.

Thanks and greetings from Switzerland !

Ronald

mcmatrix, 2021-01-21:
Hi @ronald5555 

I did some digging on this topic.
The current Jinja2 template in Klipper doesn't allow raw characters higher than 127.
Using python escape sequence like `\xf2` is also not possible in Jinja2 expression if the character code is bigger than 127.

I did some small changes in menu code.
It's currently available in my repo https://github.com/mcmatrix/klipper/tree/work-menu-unescape-20210121
Only the `menu.py` was changed.
https://github.com/mcmatrix/klipper/blob/work-menu-unescape-20210121/klippy/extras/display/menu.py

If you know the `degree` char code in your LCD charset then you can try it.

In config use `\\xXX` if inside jinja `{..}` expression otherwise `\xXX`
This way it will not throw an exception and the menu should unescape the item name.

NB! Some displays may not correctly display characters higher than 127.
For example, ST7920 was one I tested with.

ronald5555, 2021-01-22:
Hello @mcmatrix 

Thanks for your modified menu.py.

I tried it this morning and it works

Here you have my menu line with both the 7C character that points to the vertical bar (changed to ° by myself in the character generator) and the F8 character with points to the original ° symbol
name: {"FLYF407Z:%4.0f\x7c\\xf8" % (printer['temperature_sensor FLYF407ZG'].temperature)}

Of course I will not leave both in the final menu !!

and here the result:
[image]

Thanks for your work.

Do you think your menu.py file will be included in coming updates of the Klipper or will I have to check every update if the modified file is still here ?

Ronald

ronald5555, 2021-01-22:
I just see that the comment has changed my menu line.

The first character (\x7c) has a single backslash while the second character (\xf8) has a double backslash. 

In write mode I see the double backslash and in preview or published mode, the double backslash is displayed as a single backslash ??

mcmatrix, 2021-01-22:
> Do you think your menu.py file will be included in coming updates of the Klipper or will I have to check every update if the modified file is still here ?

Hi @ronald5555 
I'll test it a little bit more, maybe make a few more adjustments in code.
If it doesn't have any side-effects then I'm planning to publish it as PR
and hopefully, Kevin will merge it to the main code.

KevinOConnor, 2021-01-22:
Directly using the font at '\xf8' would not work properly on the popular st7920.  I think it would be great if we could support `~degrees~` properly in the menu items.  Instead of directly taking a string length, perhaps we could change `display.draw_text()` to return the actual length and the menu code could use that when determining if horizontal scrolling is needed?

-Kevin

mcmatrix, 2021-01-22:
> Directly using the font at '\xf8' would not work properly on the popular st7920.

You're right. Unfortunately, I also noticed this behaviour.

> I think it would be great if we could support `~degrees~` properly in the menu items. Instead of directly taking a string length, perhaps we could change `display.draw_text()` to return the actual length and the menu code could use that when determining if horizontal scrolling is needed?

I also support this approach, but I am afraid it is not an easy task. I'll try and play with some ideas.

-Janar

mcmatrix, 2021-01-24:
Please could you check this approach? #3837

ronald5555, 2021-02-15:
Hello @mcmatrix and @KevinOConnor 
Do you have any plans in merging #3837 into a coming release ?

mcmatrix, 2021-02-15:
Hi @ronald5555 
I'm a little short with time but yes I'm planning to merge it.
Need to do a couple of additional testing with other types of displays.
Anyway, you're more than welcome to also perform the test with your setup :)

BTW, when the release is planned?

mcmatrix, 2021-02-20:
Hi @ronald5555, it's merged.

KevinOConnor, 2021-03-05:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2021-03-05 by KevinOConnor

ronald5555, 2021-04-23:
Hello @mcmatrix and @KevinOConnor 
I have found a new "problem" with the display.
If You try "M117 c'est l'été" nothing is displayed due to the accented characters we use in French.
I discovered the problem with Cura that puts a M117 message at every layer change plus the filename e.g. "Layer 1 of 123 filename"
If your filename has accented characters, Klipper simply displays nothing (spaces). It would be good to either 
1. display up to the first accented character
2. replace accented characters with a ? or a space
3. strip off accented characters
4. more difficult, replace the accepted character with a standard American character eg é -> e. !!!

Perhaps you can do something ??

susisstrolch, 2021-04-23:
I also saw in the logfiles that klippy has a problem with UTF-8 characters.