Klipper3d/klipper #3799: Rpi temperature

ronald5555, 2021-01-18:
There is a small problem with the file extra/rpi_temperature.py.

If you try to use it you get an error message "get_report_time_delta undefined"

A closer look to the file showed that the 2 following lines where missing.

    def get_report_time_delta(self):
        return RPI_REPORT_TIME

I added them an the error disapeared and the Rpi temperature is reported correctly 

attached log file is of no interest for this issue
[klippy.log](https://github.com/KevinOConnor/klipper/files/5830117/klippy.log)

KevinOConnor, 2021-01-18:
I need a log showing the error.

-Kevin

ronald5555, 2021-01-20:
OK Kevin, I have reverted my modifications to show you what is happening:

With the original extra/rpi_temperature.py I get the error you will find in the klippy.log file at the last lines

  _File "/home/pi/klipper/klippy/extras/temperature_fan.py", line 157, in load_config_prefix
    return TemperatureFan(config)
  File "/home/pi/klipper/klippy/extras/temperature_fan.py", line 25, in __init__
    self.speed_delay = self.sensor.get_report_time_delta()_

With my modified file extra/rpi_temperature.py at line 39 and 40 Klipper starts correctly. You will see the correct start in the klippy3.log file again in the last lines.

Here again my modifications in the file:

**def get_report_time_delta(self):
    return RPI_REPORT_TIME**

I attach the modified file with a txt extension to be able upload.

By the way in that file you will see in klippy3.log the "timer too close" error on line 1683 but I have already created an issue for this (with no answer as of today)

Hope this will help you ..
[klippy-3.log](https://github.com/KevinOConnor/klipper/files/5840713/klippy-3.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5840714/klippy.log)
[rpi_temperature.py.txt](https://github.com/KevinOConnor/klipper/files/5840716/rpi_temperature.py.txt)

KevinOConnor, 2021-01-24:
Should be fixed now (commit c3e3b228).

-Kevin

closed 2021-01-25 by ronald5555