Klipper3d/klipper #3800: M117 suggestion

ronald5555, 2021-01-18:
If you want to display a message for a couple of seconds (e.g. "Printer homed")
you need to:

**define the following macro**
[delayed_gcode clear_display]
gcode:
  M117

**and when you display a text do the following**:
	M117 All Homed				# message
	UPDATE_DELAYED_GCODE ID=clear_display DURATION=10

**Suggestion:**

Would it be possible to add a parameter to the M117 command that will timeout the display like
      M117 All Homed  P10			# message

log file is only here as placeholder
[klippy.log](https://github.com/KevinOConnor/klipper/files/5830149/klippy.log)

Arksine, 2021-01-19:
There is no good way to parse commands from the M117 message without potentially breaking its usage for other users.    That said, if you are comfortable with jinja2, you could likely implement this locally with by overriding the M117 gcode (see the "rename_existing" option in the docs).

ronald5555, 2021-01-20:
Thank you for you answer that I completely understand.
I will either leave my menus like there are or create a new macro of my own with the 2 parameters on one line.

Thanks

closed 2021-01-20 by ronald5555