Klipper3d/klipper #3801: PT100 INA826 ignores adc_voltage

ApolloRR, 2021-01-18:
https://github.com/KevinOConnor/klipper/blob/28bca7da7741bafbbe8619da2c46892b38696191/klippy/extras/adc_temperature.py#L269

The regular PT100 INA826 works fine with only 3.3v and the function could be adjusted to use the adc_voltage instead of the hard coded "5." volts.

Hooking up the INA826 to only 3.3v is beneficial so there is no chance of pushing 5v to a non 5v tolerant pin on the IC.

I am using the SKR 1.3 and using a custom ADC table

[adc_temperature my_sensor]
#temperature1:
#voltage1:
#temperature2:
#voltage2:

Adjusting the 5. to 3.3 in adc_temperature.py also worked fine.

zellneralex, 2021-01-19:
Kevin, I would have expected that the module use the config parameter adc_voltage but the adc_temperature.py use a hard coded 5 value. Is there are reason for that?
Coding Reference:
sensor_type:
#   One of "PT100 INA826", "AD595", "AD597", "AD8494", "AD8495",
#   "AD8496", or "AD8497".
sensor_pin:
#   Analog input pin connected to the sensor. This parameter must be
#   provided.
#adc_voltage: 5.0
#   The ADC comparison voltage (in Volts). The default is 5 volts.
#voltage_offset: 0
#   The ADC voltage offset (in Volts). The default is 0.

ApolloRR, 2021-01-19:
For instance I believe Marlin uses a conversion function for the Ultimainboard V2.x

https://github.com/MarlinFirmware/Marlin/blob/a7f334387c98a0209433d98c44269fdb0356ab54/Marlin/src/module/thermistor/thermistor_21.h

KevinOConnor, 2021-01-19:
> I would have expected that the module use the config parameter adc_voltage but the adc_temperature.py use a hard coded 5 value. Is there are reason for that?

The sensor specification is based on 5V, so the internal table also uses 5V.  The `adc_voltage` parameter is used - it's applied in the `adc_temperature.py:LinearVoltage` class.

-Kevin

ApolloRR, 2021-01-21:
Thank you Kevin.

So if I understand correctly it's simply due to the fact that e3d spec says it should use 0v and 5v and therefore people don't usually hook it up to anything other than that, and therefore this isn't really a "bug".

I suppose on the Ultimainboard V2.x running the LPC176x with the INA826 the circuitry before amp is different to allow the use of 3.3v.

Makes sense.

Thanks

KevinOConnor, 2021-01-22:
If you feed the INA826 with 3.3V, hook it up to a 3.3V mcu, and don't specify "adc_voltage", then it should "just work".  I know it's confusing and perhaps there is a way to make the config more clear.

-Kevin

closed 2021-01-26 by klipper-gitissuebot