Klipper3d/klipper #3802: Servo moves uncontrolled during startup

cicinovec, 2021-01-19:
Hello,
I am observing the same issue as mentioned in issue https://github.com/KevinOConnor/klipper/issues/2821
During startup the servo moves uncontrolled, whenever the klipper connect to the printers everything works as supposed.

Unfortunately I dont have scope to check the signal but I have tested the behavior on 3 boards:
mega2560 (Megatronics) - issue present
ATmega328P (arduino uno) - issue not present
Creality 4.2.2 board - present

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5833977/klippy.zip)

KevinOConnor, 2021-01-19:
Some boards will leave a pin floating at startup, or the bootloader on the board may toggle the pins in some way which upsets the device.

One thing you could try is setting the pin state in "make menuconfig" (enable the "low level options" and you should see an entry for specifying startup values of pins).

-Kevin

cicinovec, 2021-01-20:
Hi Kevin,
Thank you, the solution works perfectly.
Maybe an idea to add this to installation manual?

closed 2021-01-20 by cicinovec

feaver12, 2021-05-04:
I am also experiencing the same problem, however i have checked gpio.h and hardpwm.c in /src/linux and both already have the changes made mentioned in a previous thread.... i have tried a rf choke which makes no difference and funnily enough the reason i did this was because i did observe the pwm signal with a scope and found the problem would 100% disappear when the scope was attached,  power cycled around 20 odd times to be sure.. and the problem returns as soon as i remove the probes and power cycled. Hence why i thought it was noise and the scope was filtering out just enough to stop the problem but as i said i fitted a ferrite bead to the servo cable amd it has made little-no difference... what else could be causing this? Seems strange that probing the servo signal pin with the scope would stop the problem. I have also made 200% sure the servo has a good gnd even grounding it at the board and power supply to be sure. Besides trying sheilded wire im lost.