Klipper3d/klipper #3805: BTT E3 Mini V2.0 with CR20-Pro LCD

nimble0, 2021-01-19:
I've gotten this combination working with Marlin (bugfix-2.0.x) by specifying the display as MKS_MINI_12864 and using this wiring:
[image]

I tried to adapt this to my klipper config and got the error `pin PB13 is reserved for spi2`:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5837918/klippy.log)

If I change the spi_bus to spi1 then I don't get an error but the LCD remains blank (but powered on):
[klippy-spi1.log](https://github.com/KevinOConnor/klipper/files/5837919/klippy-spi1.log)

KevinOConnor, 2021-01-24:
If you want to wire a uc1701 display to that board you will need to use software spi - see: https://www.klipper3d.org/Config_Reference.html#common-spi-settings

-Kevin

nimble0, 2021-01-27:
Thanks, the screen is working well now. This is the display config I ended up with:

```
[display]
lcd_type: uc1701
cs_pin: PB8
a0_pin: PB9
encoder_pins: ^PA9, ^PA10
click_pin: ^!PA15
spi_software_miso_pin: PC12
spi_software_mosi_pin: EXP1_8
spi_software_sclk_pin: EXP1_1
```

I think this probably would work with the 1.2 version of the board as well https://github.com/KevinOConnor/klipper/issues/3005

closed 2021-01-27 by nimble0