Klipper3d/klipper #3806: 'stepcompress o=6 i=-1 c=1 a=0: Invalid sequence' before 'Internal error on command:"G1"'

DefinitelyNotSam, 2021-01-20:
[1-19-21.zip](https://github.com/KevinOConnor/klipper/files/5839680/1-19-21.zip)

Print halted mid-print, Mainsail displaying 'Internal error on command:"G1"'. Digging through klippy.log (attached), I believe the issue occurred on line #26288, but I'm pretty new to Klipper.

Just started trying to reprint the same gcode to see if it will fail in the same spot. I recently swapped to an LV8729 for my extruder driver (from a TMC2209), and I'm using 128 microsteps (overkill, I realize).

Attached Zip contains gcode, klippy.log, and printer.cfg.

Appreciate any help identifying and resolving this! I was lucky it didn't 'bite' me on a large print, but it easily could have!

Thanks!

DefinitelyNotSam, 2021-01-20:
Reprint of the same gcode was successful.

Sineos, 2021-01-20:
May I ask why you switch a TMC2209 (1.7A RMS / up to 256 mu-steps) for a less powerful LV8729 (1.5 A RMS / 128 mu-steps).
I stumbled across people doing this a few times and never understood why.

DefinitelyNotSam, 2021-01-20:
> May I ask why you switch a TMC2209 (1.7A RMS / up to 256 mu-steps) for a less powerful LV8729 (1.5 A RMS / 128 mu-steps).
> I stumbled across people doing this a few times and never understood why.

I actually did it because I only have 4x TMC2209's on hand, and recently took the extruder driver for a second Z axis. The LV8729 seemed like a worthy replacement for an extruder stepper out of my spare drivers.

KevinOConnor, 2021-02-08:
FYI, the error indicated (an "internal error in stepcompress") is usually an indication of a bug in the software.  Unfortunately, I haven't been able to find any leads as to what conditions would cause the software to do that.

-Kevin

DefinitelyNotSam, 2021-02-10:
I appreciate the info, Kevin. I've since printed multiple 12+ hour prints with no issues, same version of Klipper.

I'm happy to chalk this one up to radiation bit-flip at this point and get this off your list.

closed 2021-02-10 by DefinitelyNotSam