Klipper3d/klipper #3808: Hardware PWM not working on FLYF407ZG

ronald5555, 2021-01-20:
I would like to switch my case light output from software PWM to hardware PWM

Purpose: with software PWM the light is blinking instead of dimming due to the high cycle time (normal).
With hardware PWM and a capacitor, the higher cycle time will allow dimming of the case light.

Error:
when setting the "hardware_PWM" parameter to True (see line nb 1239 of log file), I get the following error: 

 File "/home/pi/klipper/klippy/mcu.py", line 203, in _build_config
    self._pwm_max = self._mcu.get_constant_float("PWM_MAX")

My mcc board (where the case light is connected) is a FLYF407ZG with a stm32f407xx processor

Reverting the "hardware_PWM" parameter to False makes everything work OK but in software mode.

Thanks in advance for a hint.

Ronald
[klippy-2.log](https://github.com/KevinOConnor/klipper/files/5840843/klippy-2.log)

ShohninDmitriy, 2021-01-23:
All STMs do not support hardware_PWM in the firmware

KevinOConnor, 2021-01-24:
As indicated hardware pwm hasn't been implemented on the stm32 boards.  You don't need hardware pwm though - just use software pwm and set the cycle time to .010 (10ms).  If desired, on a fast micro-controller like the stm32f407 you can even switch faster than 10ms.

https://www.klipper3d.org/Config_Reference.html#output_pin

-Kevin

jorkkiporkki, 2021-01-26:
As much as I love Klipper I also love the Sunon Vapo fans. Unfortunately they do not work with software PWM as they have an autorestart function which gets confused. They start working really well about above 20 kHz though. It would be very nice to get the hardware PWM to work at some point on the STM32F1 etc. Unfortunately I don't have any ARM experience, so it would probably take me a while to try and implement it myself. I assume there are pin and timer restrictions based on what is available on different processor versions and what timers Klipper uses internally. One further complication is that many boards have badly designed fan outputs with the CPU driving a huge Mosfet with big gate charge values directly, which is sub-optimal for fast operation. That does not bother me as I have my own electronics that I use. I just need a pin, which I can steal from the original fan output, BLTouch interface etc. If this is not possible for some reason, I'm contemplating designing a PWM-to_DC converter, which would be optimal for all fans, but probably unneccessary if there is hardware PWM available.

marioy0, 2021-02-02:
ok thank you for the information. can you recommend another 4020 fan with software pwm, that is similar to sunon ?

closed 2021-03-04 by github-actions[bot]