Klipper3d/klipper #3811: Accepting NaN values causes unexpected results

cadriel, 2021-01-20:
It's currently possible to send `NaN` in console, as per below;

```
SET_HEATER_TEMPERATURE HEATER=extruder TARGET=NaN
```

In this case, Klipper starts heating - and will error once it exceeds the configured max temp.

Secondarily, Javascript reject's a `NaN` in its `JSON.parse` - causing many more issues client side. For example, in Fluidd's case the temperature value stop updating, so the user is unaware of what his / her heater is doing.

Can include a log here if you like, but suspect this is clear enough.

@Arksine @KevinOConnor

cadriel, 2021-01-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5845487/klippy.log)

bevanweiss, 2024-08-14:
There's probably a few special cases here...
NaN, +Inf, -Inf
These are 'valid' floating point values, but do not represent a numeric value.  Which also means that they can't be captured by JSON.

I would argue where ever JSON stringify-ing occurs, these should be converted as:
NaN => null
+Inf => Max.Float (i.e. +1.70141183E+38)
-Inf => Min.Float (i.e. -1.70141183E+38)

Hopefully the other side would then appropriately handle these (the max/min should be fine... null would probably need special handling on retrieving from JSON).

On the Klipper internal, I think things should handle NaN 'correctly' (in addition to the +Inf / -Inf values).
+Inf / -Inf are probably fine, as they typically work in comparisons.
But all comparisons with NaN will fail, so if the NaN is the target temperature, and the logic is 'if temperature > target then turn off' this will fail, and will always be on.
If the logic was instead 'if temperature < target then turn on' it should have been 'fine' (it would never turn on).

An explicit test for whether a setpoint is NaN and performs the 'safest' action in that event is probably prudent... but it does seem like a lot of work.

ChasonDeshotel, 2025-01-08:
> I would argue where ever JSON stringify-ing occurs, these should be converted as:
> NaN => null
> +Inf => Max.Float (i.e. +1.70141183E+38)
> -Inf => Min.Float (i.e. -1.70141183E+38)

I would argue that's invalid input. Why should Klipper know anything about JavaScript primitives? Until we have Klipper.JS it would be awkward to include logic to translate those. Besides, NaN, +Inf, and -Inf aren't valid JSON anyway

I think a good, easy solution would be to use the `allow_nan` argument that the `json.dumps` method provides

`jmsg = json.dumps(data, separators=(',', ':'), allow_nan=False)`

> allow_nan ([bool](https://docs.python.org/3/library/functions.html#bool)) – 
>If False, serialization of out-of-range [float](https://docs.python.org/3/library/functions.html#float) values (nan, inf, -inf) will result in a [ValueError](https://docs.python.org/3/library/exceptions.html#ValueError), in strict compliance with the JSON specification. If True (the default), their JavaScript equivalents (NaN, Infinity, -Infinity) are used.

I couldn't test this because Klipper crashes for me when I send `SET_HEATER_TEMPERATURE HEATER=extruder TARGET=NaN`. I get `Klipper state: Shutdown` and can't recover without restarting the service from the terminal

KevinOConnor, 2025-02-02:
I'm going to close this as we no longer use the Klipper issue tracker. The few remaining issues that are open can lead to confusion.

-Kevin

closed 2025-02-02 by KevinOConnor