Klipper3d/klipper #3812: rescheduled timer in the past - multi mcu build

Zakm0n, 2021-01-20:
So, I'm getting rescheduled timer in the past mcu shutdown problems on my dual mcu build. The board that has only my Z axis steppers is shutting down for no apparent reason when in the middle of a perimeter. I'm not using any mesh leveling, as I have the 3 stepper tilt adjustment set up on this board. Also have accelerometer configured input shaper. Log file is zipped, as it was too big to attach.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5845687/klippy.zip)

jakep82, 2021-01-21:
From your log:

```
Rescheduled timer in the past
Transition to shutdown state: MCU 'mcu' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
```

This indicates your primary MCU is shutting down, not MCU Z.  You appear to be using a highly geared extruder paired with an 8 bit AVR.  At ~2200 steps per mm for your extruder, you'll exceed the safe limits of your primary MCU somewhere around 20-30mm/s for retractions.  You can avoid this issue by reducing the microstep setting on your extruder.

Zakm0n, 2021-01-21:
Huh, alright, guess I was mistaken. I am using a highly geared extruder. My Z mcu is 32 bit, so maybe I'll move the extruder over to it since E and Z never run at the same time... Thanks!

closed 2021-01-21 by Zakm0n