Klipper3d/klipper #3815: Klipper activates all fans when Pi shuts down

Roycinger, 2021-01-21:
Hi folks,

on all my Klipper driven printers, I noticed that all controlled fans are driven at 100% as soon as the host / Pi shuts down.
They only stop when I flip the physical power switch and cut power to the entire system.
This is somehow undesireable to me.
Is there a possibility to suppress this behaviour?

For good measure: 
[klippy (7).log](https://github.com/KevinOConnor/klipper/files/5851946/klippy.7.log)

Hywelmartin, 2021-01-21:
(https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#fan

[image]

Roycinger, 2021-01-21:
Thanks. If the default is 0, the fans shouldn't run then if I understand it correctly. Though, they do.
Nevertheless, setting this parameter explicitely solved the fans spinning on Pi shutdown.
Also the illumination if the LEDs on my smart effector turn on, which I can't suppress with shutdown_speed

Hywelmartin, 2021-01-21:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#output_pin

[image]

Roycinger, 2021-01-21:
The illumination LEDs on the smart effector dont have an assigned pin. They turn on whenever the hotend turns on:
[image]
I dont see the hotend heating up after the Pi shutdown, nevertheless

jakep82, 2021-01-21:
The part cooling fan has a default shutdown_speed of 0.  Most other fans have a default value equal to max_power.  This is explained in the documents.

https://www.klipper3d.org/Config_Reference.html#heater_fan

Roycinger, 2021-01-21:
Alright, that makes it totally clear! Thanks for the assistance!

closed 2021-01-21 by Roycinger