Klipper3d/klipper #3816: !! Move out of range: 183.000 120.000 -10.006 [2.500]

dertbv, 2021-01-21:
When trying to print i am getting the error message after the probe lifts from the bed. Installed fluid over the weekend, This was working yesterday before the a pull last night. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5852320/klippy.log)
[printer(cfg).log](https://github.com/KevinOConnor/klipper/files/5852324/printer.cfg.log)
[mainsail(cfg).log](https://github.com/KevinOConnor/klipper/files/5852328/mainsail.cfg.log)

dertbv, 2021-01-21:
i have narrowed down my issue to my prime line. If i rem it out I can print with no problem.

[gcode_macro PRIME_LINE]
gcode: 
    M117 Prime Line
    G92 E0 ;Reset Extruder
    # move z axis 
    G1 Z2.0 F3000 ;Move Z Axis up
    # move to prime position 
    G1 X20 Y30 Z0.28 F5000.0 ;Move to start position
    G1 X20 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
    G1 X22 Y200.0 Z0.28 F5000.0 ;Move to side a little
    G1 X22 Y50 Z0.28 F1500.0 E30 ;Draw the second line
    G92 E0 ;Reset Extruder
    G1 Z2.0 F3000 ;Move Z Axis up

jakep82, 2021-01-22:
Your default_parameter_z in your PRIME_LINE macro is -20.  I'm not sure what exactly your passing in the gcode, but that doesn't seem right to me.

```
[gcode_macro PRIME_LINE]
gcode = 
	SAVE_GCODE_STATE NAME=STATE
	G91
	G1 Z{Z} E{E} F900
	RESTORE_GCODE_STATE NAME=STATE
	RESUME
default_parameter_e = 2.5
default_parameter_z = -20
```

Arksine, 2021-01-22:
You are running a fork of Klipper.  We can't help troubleshoot this unless you can reproduce with pristine code from this repo.

dertbv, 2021-01-22:
That was the issue.  The numbers you are referring to were part of the advanced resume macro.  there was a space before the line that was causing it.  All fixed.

I thought Fluid was just the front end for klipper not a fork of it?

Arksine, 2021-01-22:
> I thought Fluid was just the front end for klipper not a fork of it?

You are correct, it is not a fork.  I apologize, I made a mistake when I searched the commit hash.  When I attempt to troubleshoot issues it is always the first thing I do so I can get an idea of what code is running on the machine.  Your version of Klipper is from this repo, so you are good.

closed 2021-02-13 by dertbv