Klipper3d/klipper #3819: Klipper service quietly crashes.

mattswest, 2021-01-22:
There doesn't seem to be anything written at the end of the log file when it crashes.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5853104/klippy.log)

Octoprint gives the error below. 

```
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,154 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Jan 21 20:36:22 octoprint-white octoprint[1916]: Traceback (most recent call last):
Jan 21 20:36:22 octoprint-white octoprint[1916]:   File "/home/matt/OctoPrint/lib/python3.8/site-packages/octoprint/util/comm.py", line 3831, in _readline
Jan 21 20:36:22 octoprint-white octoprint[1916]:     ret = self._serial.readline()
Jan 21 20:36:22 octoprint-white octoprint[1916]:   File "/home/matt/OctoPrint/lib/python3.8/site-packages/octoprint/util/comm.py", line 6455, in readline
Jan 21 20:36:22 octoprint-white octoprint[1916]:     c = self.read(1)
Jan 21 20:36:22 octoprint-white octoprint[1916]:   File "/home/matt/OctoPrint/lib/python3.8/site-packages/serial/serialposix.py", line 595, in read
Jan 21 20:36:22 octoprint-white octoprint[1916]:     raise SerialException(
Jan 21 20:36:22 octoprint-white octoprint[1916]: serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,157 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,175 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,191 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,274 - octoprint.plugins.printoid - INFO - Could not load image from url
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,275 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Offline (White)
Jan 21 20:36:22 octoprint-white octoprint[1916]: 2021-01-21 20:36:22,691 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Offline (White)
Jan 21 20:36:24 octoprint-white octoprint[1916]: 2021-01-21 20:36:24,525 - octoprint.plugins.tracking - INFO - Sent tracking event print_failed, payload: {'origin': 'local', 'file': '026313fba492efe6ac3ccbd43c6327e36f16d1d0', 'elapsed': 373, 'reason': 'error', 'commerror_text': "SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831"}
```

This seems to be because the link at /tmp/printer is broken.

Which I guess is because the process has exited even though the service remains running.

```
● klipper.service - LSB: Klipper daemon
     Loaded: loaded (/etc/init.d/klipper; generated)
     Active: active (exited) since Mon 2021-01-18 21:08:50 EST; 3 days ago
       Docs: man:systemd-sysv-generator(8)
    Process: 1171177 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
```

Am I correct in thinking this isn't an issue with Klipper losing communication with the printer, but with Octoprint losing connection with Klipper? I checked for USB disconnects in dmesg output but didn't see any. I read [this](https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors) but it doesn't seem to be what I'm experiencing. Is this the expected behavior if the USB connection drops?

The LCD remains on and frozen, but the heaters and motors get turned off.

When I restart the Klipper service and reconnect OctoPrint, I see the following.

```21:33:52 Can not update MCU 'mcu' config as it is shutdown
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
21:33:52 Can not update MCU 'mcu' config as it is shutdown
```

I generated a load graph from the Klipper log.

[image]

The first section is a successful print from earlier in the day, followed by some idle time. The second section starts with me updating the config via OctoKlipper triggering a restart, followed immediately by the second print which crashed.

I also tried to extract shutdown information from the logs but there didn't seem to be any.

I'm not sure what else I can do to troubleshoot this without assistance; please let me know if there is anything I can do to help solve this problem.

KevinOConnor, 2021-01-22:
If the host "klippy process" exited without any logs it generally indicates a severe system problem.  (For example, corrupted sd card, voltage irregularities, etc.)  If you're familiar with Linux then you could try looking through the system logs at the time to see if they report any issue.  Otherwise, without logging information from the event, there's not much we can do to assist.

-Kevin

mattswest, 2021-01-28:
Just had this happen again.

I've looked through all the system logs and found nothing out of place. The excerpt above is from the systemd journal, and the kernel message buffer shows nothing. I also checked the auth logs and saw nothing at that time.

I noticed some launch options for klippy.py; could running in verbose mode, or without logging to file possibly show more information when it crashes? Is it possible to get more logging from the Python interpreter? I'm not too familiar with Python, '-d' perhaps?

I've had this happen on two identical setups.

Ender 3 with BTT SKR Mini E3 V2.0
Rock64 1GB SBC (RK3328 SOC)
Armbian Focal (5.9.x Kernel)
Sandisk SD Card

closed 2021-02-26 by github-actions[bot]

mattswest, 2021-03-12:
Still happening. If I didn't already mention this, part of the reason this issue is difficult to troubleshoot is that it only happens rarely, and usually toward the start of a print.

I switched out the SBC with a different, but identical setup; different power adapter, board, and SD card.

Honestly, there's nothing anywhere in this process to indicate that there is anything wrong with anything other than the software; there are no systems errors to be found. I really don't see how you can justify the statement:

> If the host "klippy process" exited without any logs it generally indicates a severe system problem.

Is this meant to be a semantic arguement? Klippy wouldn't "exit" without any logs? I don't assume it's "exiting", it seems to be just crashing. 

To me, it just seems like you are summarily discounting the possibility that there could be _anything_ wrong with _your_ software. 

I was really excited about this project at one point, even considered donating, but the attitude I've seen from you has discouraged me from doing so, and has turned my excitement into disappointment that this is not a project that can be relied on.

I've read many of the issue reports on here, and it seems your go-to reaction is to assume the fault lies with the end-user. I also read the pull request you wrote saying that "95% of new github issues are really "support queries" - tickets from users looking for assistance"

This is a distinction you've created in your mind to avoid dealing with the issues people have with your software.

If this software is crashing, without any indication why, when everything else on the system is running normally, and no amount of troubleshooting can narrow down the problem to anything outside of the software, then it _is_ a bug until it can be determined otherwise.

I don't understand why a developer would take this attitude. Have you tested every combination of every version of Klipper, with every kernel, every python version, every distinct combination of hardware for the computer and controller? No, that's the benefit of feedback from end-users. If your reaction is to ignore all of the feedback because it's too onerous, then you, and more importantly your software, don't get that benefit.

It's possible that the most effective thing you could do to reduce the amount of "github issues (that) are really support queries", is to not assume that every difficulty they have is neither your fault nor your problem.