Klipper3d/klipper #3822: using rotation_distance

jornada812, 2021-01-22:
In accordance with the recommendations, I decided to fix the config and use rotation_distance. Motors 0.9, belt GT2. The calculated value is 32, but with it my delta is incorrectly positioned. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5855642/klippy.log)
Did I calculate incorrectly?

Hywelmartin, 2021-01-22:
for 16 toothed it's 32

https://github.com/KevinOConnor/klipper/blob/master/docs/Rotation_Distance.md

jakep82, 2021-01-22:
Did you configure full_steps_per_rotation?  The default is 200 which corresponds to a 1.8 degree stepper motor.  It's impossible to provide any more help without a log showing your configuration that includes rotation_distance.

https://www.klipper3d.org/Config_Reference.html#stepper

rizwansarwar, 2021-01-23:
I have an ET4, similar to ender 3, i updated the config to start using rotation_distance. My voron test cube comes 0.1mm short or X and 0.16 short on Y. Z is exactly 30 as expected. How do I tune/adjust  this to compensate for this situation?

rizwansarwar, 2021-02-02:
Anyone?

jakep82, 2021-02-02:
@rizwansarwar This isn't the place to ask general printer tuning questions.

https://www.klipper3d.org/Contact.html

jakep82, 2021-02-02:
Klipper and Voron each have Discords where you can ask this question and get a response.

closed 2021-03-04 by github-actions[bot]