Klipper3d/klipper #3823: Problems with changing from TMC2208 to TMC2209

icy1975, 2021-01-22:
System: Radds1.6 with arduino due and TMC2208 in Uart Mode and klipper 0.91

I tried to change to TMC2209, but i have some problems, First the  TMC config for the 2209 on Y axis
[tmc2209 stepper_y]
uart_pin: ar38
uart_address:0  ( if tried alles values from 0 -3)
microsteps: 16
interpolate: True
run_current: 1.0
hold_current: 0.7
#sense_resistor: 0.110
stealthchop_threshold: 0

the uart does not connect to the driver. And if this section and the driver are in place the thermistors play crazy and shows 0 and "nothing"

If i remove the TMC section, the thermistors are working normally.
In the terminal i am getting with the 2209section ecv: // Unable to read tmc uart 'stepper_y' register IFCNT

Maybe somebody has an idea, i have running the TMC2208 on this machine and the 2130 on another printer, but with the 2209 i am really desperate. I ve tried 4 different TMC2209 , always the same, and if i change to mode to standalone, the driver working normally, the Setting for the uart pin is correct.

Best regards

Chris

ShohninDmitriy, 2021-01-23:
Delete **uart_address:0 ( if tried alles values from 0 -3)** , uncomment **#sense_resistor: 0.110**

icy1975, 2021-01-23:
Thank you for your answer. I ve tried both, but nothing worked.i can not connect with uart to the drivers
I do not understand why the thermistor show nothing when i put the 2209 in the Radds. If i switch back to 2208 and change the firmware setting everything is working fine.

closed 2021-02-22 by github-actions[bot]