Klipper3d/klipper #3825: FR: Option for klipper.log to be in JSON format

brotherdust, 2021-01-22:
I want to export the statistics present in `/tmp/klipper.log` real-time. To do that, I'd need to write a grok filter (or a custom parser) of some sort to turn it into structured data that my exporter can understand.

Obviously, that's not a very elegant way to solve the problem and the result would be quite brittle.

The ideal would be to have an option to output the logs in JSON format. That way I can just tell Telegraf `tail -f /tmp/klipper.log` and problem solved.

I realize that using the subscribe function of the new API is an option as well, but it's ... not clear to me how I can get the equivalent output that is present in `klipper.log`.

I reviewed the logging code and it doesn't seem like a huge stretch to just serialize log output into JSON instead of the default format; however, as I possess merely _adequate_ Python skills, contributing actual code intimidates me somewhat.

So, my question to you, @KevinOConnor, is this: would you be willing to consider implementing this? If not, would you be willing to provide a bit of guidance or suggestion as to how _you_ would go about implementing so as I can do it myself and contribute it back?

Thanks in advance!

KevinOConnor, 2021-01-24:
I recommend you use the API server: https://www.klipper3d.org/API_Server.html .  (And perhaps use moonraker if you want to query over http).  Most of the information in the log is already available via the API server, and the information that is missing is likely to be added in the near future.

-Kevin

brotherdust, 2021-01-24:
@KevinOConnor thanks for the guidance here. Much appreciated.

closed 2021-01-24 by brotherdust

brotherdust, 2021-01-27:
For anyone running across this later, I found a project that does a pretty good job of turning the Klipper API into a web API. Very nice!
[Moonraker](https://github.com/Arksine/moonraker)