Klipper3d/klipper #3836: Bug/feature Heater don't turn off if lcd is at temp setting

Hywelmartin, 2021-01-24:
Yesterday I heated up my bed from the LCD when I was removing a print.. and left it as it should timeout in set 10 min.. it didn't..

[image]
[image]
[klippyheater_on_forever.zip](https://github.com/KevinOConnor/klipper/files/5862101/klippyheater_on_forever.zip)

mcmatrix, 2021-01-25:
Hi @Hywelmartin 
I tested this scenario on my printer, having following idle settings
```ini
[idle_timeout]
#timeout: 360
timeout: 120
gcode:
    M117 HOME/ HEAT!!!
    TURN_OFF_HEATERS
    M84
```
1. Test 
- Printer LCD is at the home screen.
- Set bed temperature to 35C from Repetier-Server web UI
- Homed the printer and waited.
- After approx. idle time the bed temp target was turned to 0

2. Test 
- Set bed temperature to 35C from Repetier-Server web UI
- Homed the printer and in LCD activated the Temperature menu and set the cursor to Bed temp (like in your image).
- After approx. idle time the bed temp target was turned to 0

So in both cases, the idle timeout was working and setting heaters target to 0

Hywelmartin, 2021-01-25:
I can do this on both of my printers...

from LCD set temp and leave the cursor like the picture... 

[klippy (2).log](https://github.com/KevinOConnor/klipper/files/5869239/klippy.2.log)

closed 2021-01-28 by Hywelmartin