Klipper3d/klipper #3841: Config_Reference.md using non-markdown tags

chorca, 2021-01-25:
Small issue.
Was reading the [menu] section of Config_Reference.md and noticed there were a few missing bits around the menu actions.
Looks like some tags snuck in that aren't supported in Markdown:

**File:**
```
List of actions for menu element:
* menu.back(force, update): will execute menu back command, optional
  boolean parameters <force> and <update>.
  * When <force> is set True then it will also stop editing. Default
    value is False
  * When <update> is set False then parent container items are not
    updated. Default value is True
* menu.exit(force) - will execute menu exit command, optional boolean
  parameter <force> default value False
  * When <force> is set True then it will also stop editing. Default
    value is False
```

**Render:**
List of actions for menu element:

- menu.back(force, update): will execute menu back command, optional boolean parameters and .
   - When is set True then it will also stop editing. Default value is False
   - When is set False then parent container items are not updated. Default value is True
- menu.exit(force) - will execute menu exit command, optional boolean parameter default value False
   - When is set True then it will also stop editing. Default value is False

I think the <> are being interpreted as tags and aren't showing. Not sure what format they were intended to be in so can't suggest much, but a quick fix to make the doc a bit more readable.

mcmatrix, 2021-02-23:
I'll check it.

KevinOConnor, 2021-05-29:
This should be fixed now (commit 624d94f6).

-Kevin

closed 2021-06-06 by github-actions[bot]