Klipper3d/klipper #3842: Need a way to control bed heater ramp up speed

michelluccote, 2021-01-25:
I run a CR10 Max and do long prints. To avoid power outage problems I run it on a UPS that is close to it's limit in max power output. I need to ramp up the bed heater but not at maximum power and then heat up the nozzle. The UPS can easily support the bed after it is warm but will trigger alarms and sometimes shut down during warm up. Is there a way to do warm up at 80% instead of 100% using Klipper?

simonkuehling, 2021-01-27:
Hi @michelluccote, the [heater_bed] config section uses the same heater parameters that are documented in the [extruder] section (see https://www.klipper3d.org/Config_Reference.html#extruder)
```
max_power: 0.8
#   The maximum power (expressed as a value from 0.0 to 1.0) that the
#   heater_pin may be set to. The value 1.0 allows the pin to be set
#   fully enabled for extended periods, while a value of 0.5 would
#   allow the pin to be enabled for no more than half the time. This
#   setting may be used to limit the total power output (over extended
#   periods) to the heater. The default is 1.0.
```
should do the trick for limiting your heated bed to 80%!

michelluccote, 2021-01-27:
That is great and should solve my problem. I'm assuming here this will limit the heater bed PWM signal to a maximum duty of max_power as oppose to still allow it to go 100% but only for shorter periods of time. Max power even for a few seconds for me will cause the UPS to struggle. If my assumption is correct then that will work. 

Thank you for the quick reply.

simonkuehling, 2021-01-27:
That is correct, this will directly limit the duty cycle to a maximum of max_power

michelluccote, 2021-01-28:
I have done my testing and I'm happy to report that it is perfect. Thank you for the great support.

closed 2021-01-28 by michelluccote