Klipper3d/klipper #3844: TUNING TOWER COMMAND math issues

KaizenGrit, 2021-01-26:
I'm sorry if this is something I am too much of a noob to figure out, but I have done my best, searched quite a bit, and started a reddit thread.  Please help me with the correct command to use Tuning Tower Command to produce temp changes at band change correctly.  I have tried "TUNING_TOWER COMMAND="SET_HEATER_TEMPERATURE HEATER=extruder" PARAMETER=TARGET START=250 FACTOR=-1 BAND=10" and expected to see it drop by 10deg every band of 10mm.  It drops 15deg after the first band of 10mm then produces the desired outcome for every band thereafter- the problem being that now none of the values are the ones I desired.  There must be something in the math that produces this additional 5deg after the first band.  I read something about "floor" in the math when using "band", but I do not understand what it is, nor did I find further documentation. It would be great to specify something like floor=1.4mm and that is ignored in the math.  
Anyways.. I am perfectly happy to just get easy predictable temp changes after bands without using Cura GCODE mod for this or editing each GCODE file myself.  
Thanks!

[klippy.log](https://github.com/KevinOConnor/klipper/files/5871018/klippy.log)

Sineos, 2021-01-26:
The command is explained here: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md and is following this formula:
`value = start + factor * ((floor(z_height / band) + .5) * band)`

The behavior you are seeing is due to the +0.5 correction in the formula. This is necessary, otherwise the formula would "miss the last band", e.g. at Z=250 in your case. 

Using the values of your command, this means you will already get a 5 °C drop at Z=0 and then an additional drop of 10 °C at Z=10.
To counter the effect you will have to modify your command to start at 255 °C. This will then give you the initially desired starting value of 250 °C at Z=0 and a drop of 10 °C there after.

KaizenGrit, 2021-01-26:
That makes perfect sense to me now. Thank you.

closed 2021-01-26 by KaizenGrit

blalor, 2021-07-10:
I ended up here trying to understand the change proposed in #4439.  I'll accept that some additional math needs to be done on the part of the user, but this really needs to be documented well.  For @KaizenGrit's example a 10° drop every 10mm (factor: -1.0) you need to add 5 to START, but if you want to do a 5° drop every 10mm (factor: -0.5) you need to add 2.5.  It seems like this is something the `TUNING_TOWER` command could correct for automatically.

psomero, 2021-08-06:
> The command is explained here: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md and is following this formula:
> `value = start + factor * ((floor(z_height / band) + .5) * band)` [...]

I am assuming that floor() is a function and not a parameter we're passing?

ubarbaxor, 2021-09-13:
> I am assuming that floor() is a function and not a parameter we're passing?

It is. For reference, floor() takes the nearest-lower-round value. 1.0->1.9999(...) floor to 1.0, etc..