Klipper3d/klipper #3845: Small & self correcting intermittend layer-shifts

Sineos, 2021-01-26:
I'm having an issue that is driving me completely crazy: 
On some prints I'm experiencing layer-shifts. They are:

- Around .1 mm
- Concentrated on mainly the Y-axis
- Correcting themselves after a few layers

OK:
[image]
 
Shifted:
[image]

The attached log contains both prints:
- The last print is the shifted one (printing 7x these clamps)
- The print before the last one is the un-shifted

Hardware:
- TMC5160 on SKR1.4T
- Big copper heat sinks on the drivers
- 120 mm fan cooling the drivers 

Things I tried w/o result:
- Swapping X and Y drivers
- Disabling PA
- Disabling Input-Shapers

I'd really appreciate any pointer to continue investigating this issue.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5872710/klippy.log)

jakep82, 2021-01-26:
Have you tried reducing accel to decel?  7000 is very aggressive, especially if it's a bedslinger.

Sineos, 2021-01-27:
> Have you tried reducing accel to decel? 7000 is very aggressive, especially if it's a bedslinger.

You are absolutely correct. In fact this was a left-over from my Input-Shaper tests (forgot to set it back)
My regular settings are:
```
max_accel: 3000
max_accel_to_decel: 2000
```

In fact the same effect also happens with these regular settings.
It is also not depending on the slicer (tried Cura, Prusa and S3D) and it does not happen regularly at the same height. The mechanical components are running smoothly, i.e. no bonding, low breakaway-torque etc.

jakep82, 2021-01-27:
The only other thing I can think of is a slipping set screw on a pulley.

Sineos, 2021-01-27:
Thanks for the idea. Checked but unfortunately no. Would have been too easy. 😉 

I also would have been surprised if it was a set screw. Since doing a lot of resonance / vibration tests, I have secured each and every frigging screw Loctite. 

This is driving me nuts. The last few prints came out perfect. Surprisingly it also happens more often with layer heights of .2 and less with .1 or .3 (or so at least my subjective feeling)

BlackStump, 2021-01-29:
long shot - try setting your layer heights to multiples of your step distance

Sineos, 2021-01-29:
> long shot - try setting your layer heights to multiples of your step distance

Thanks for the input @BlackStump. In fact, I'm already doing so.
I have created an Excel file to calculate it for the new Rotation Distance. I have uploaded it here as Google spread-sheet (you need to make a copy to modify):

https://docs.google.com/spreadsheets/d/1P2lPFVESzimx-DUtjT575pA9t00lrSTNjlXnr_3B0Yk/edit?usp=sharing

It seems to give sensible values. Anyway I'd appreciate a review.

BlackStump, 2021-01-30:
With such a high res Z your layer height should be multiples of Full Steps
ie 53 full steps = 0.1980875
Curious what hardware you have on the Z to equal such a odd number?

Sineos, 2021-01-30:
> Curious what hardware you have on the Z to equal such a odd number?

Lead-screw: TR8x1.5
Stepper: 0.9°

Nominal rotation distance would be 1.5 but I did extensive measurements with a 50 mm dial gauge on various heights. So corrected the nominal value to 1.495. According to my measurements and the printing results, it is repeatable and an improvement. 

> With such a high res Z your layer height should be multiples of Full Steps
> ie 53 full steps = 0.1980875

I do not understand, how you arrive at 53 full steps. Would you please explain.

My reasoning is as follows:

```
step_length = rotation_distance / full_steps_per_rotation / microsteps = 1.495 / 400 / 64 =  0.0000583984375
layer_height_multiple = layer_height / step_length = 3424.749163879599
floor(layer_height_multiple) = 3424
ceil(layer_height_multiple) = 3425
even_multiple_layer_height_lower =  floor(layer_height_multiple) *  step_length = 3424 *  0.0000583984375 = 0.19995625
even_multiple_layer_height_upper =  ceil(layer_height_multiple) *  step_length = 3425 *  0.0000583984375 = 0.2000146484375
```

When omitting the microsteps, i.e. taking only 400 full steps:
```
step_length = rotation_distance / full_steps_per_rotation = 1.495 / 400 =  0.0075333984375
layer_height_multiple = layer_height / step_length = 26.5484431308496
floor(layer_height_multiple) = 26
ceil(layer_height_multiple) = 27
even_multiple_layer_height_lower =  floor(layer_height_multiple) *  step_length = 26 *  0.0075333984375 = 0.195868359375
even_multiple_layer_height_upper =  ceil(layer_height_multiple) *  step_length = 27 *  0.0075333984375 = 0.2034017578125
```

BlackStump, 2021-01-30:
With such a high resolution Z, microstepping brings very little to the table above full steps in your case, maybe less noise at the expense of your mcu load. I just see it as a diminishing return.
But of course it is entirely your choice to do as you see fit.

Sineos, 2021-01-31:
> With such a high resolution Z, microstepping brings very little to the table above full steps in your case, maybe less noise at the expense of your mcu load.

Agreed. Will most likely not make much of a difference in terms of precision but it will end up on the more robust full-steps instead of micro-steps.

I'm in the process of modifying my Ender 5 from bowden to direct drive. So most likely the printer will anyway start behaving differently (and create new challenges to cope with).

Have not been able to find a reason for this intermittent behavior but thanks a lot for the input here. Appreciated.

closed 2021-01-31 by Sineos