Klipper3d/klipper #3849: MCU Port: License Compatibility

brotherdust, 2021-01-26:
@KevinOConnor ,
I'm considering porting Klipper to the Kendryte K210 MCU. [The documentation for MCU porting](https://www.klipper3d.org/Code_Overview.html#porting-to-a-new-micro-controller) mentions that _"All 3rd party code needs to be GNU GPLv3 compatible."_.

[kendryte-standalone-sdk](https://github.com/kendryte/kendryte-standalone-sdk) is Apache 2.0 licensed. [This Wikipedia article](https://en.wikipedia.org/wiki/Apache_License) states that _"The Apache Software Foundation and the Free Software Foundation agree that the Apache License 2.0 is a free software license, compatible with the GNU General Public License (GPL) version 3,[7] meaning that code under GPLv3 and Apache License 2.0 can be combined, as long as the resulting software is licensed under the GPLv3.[8]"_.

Before I even start trying (and I know I'm in WAY over my head, but I learn best this way), I just wanted to get a sign-off from you saying this is OK.

Please advise.
Thanks!

brotherdust, 2021-01-28:
Eh. I'll take the language in the wording in the MCU porting guide literally.

closed 2021-01-28 by brotherdust

KevinOConnor, 2021-01-28:
FYI, I interpret the Apache 2.0 license as compatible with the GPL.  Indeed, there are several Apache2 libraries included in Klipper's lib/ directory.

-Kevin