Klipper3d/klipper #3850: MainsailOS Klipper doesn't boot on printer

andriusmot, 2021-01-26:
Hello guys, I've been following this tutorial on how to install klipper on Sapphire plus printer using Robin Nano 1.2 board. I did installed MainsailOS on RaspberryPi, no problems there. After building firmware with mainsailOS and trying to boot it on the printer, nothing happens. I plug SD card in and Im getting only "Booting", nothing happens after. I tripple checked all the steps in tutorial and Im pretty sure I'm following it correctly. Please let me know if you have any advise on how to solve it, thanks.

https://3dprintbeginner.com/how-to-install-klipper-on-sapphire-plus/

[klippy (1).log](https://github.com/KevinOConnor/klipper/files/5876360/klippy.1.log)

raymondh2, 2021-01-27:
The serial port you specified in the [mcu] config does not exist. Are you sure it is plugged in and that is the correct serial port?

andriusmot, 2021-01-27:
> The serial port you specified in the [mcu] config does not exist. Are you sure it is plugged in and that is the correct serial port?

Could you expand on this a bit more? When compiling klipper I have 3 serial port options, tried all of them but still result is the same

Daali, 2021-01-27:
ls /dev/serial/by-id/*  

display's what?

andriusmot, 2021-01-27:
> ls /dev/serial/by-id/*

in log file or where?

andriusmot, 2021-01-27:
These are exactly same things that are happening to me, I'm not inclined enough to understand everything what's goin on there I'm afraid, but topic concluded with support being made for this issue, so maybe someone can take another look and give me some pointers please? 

https://github.com/KevinOConnor/klipper/issues/2397

andriusmot, 2021-01-27:
found it
idk if its my particular case or not, but it works. After compiling firmware with KIAUH I ran this script
`~/klipper/scripts/update_mks_robin.py out/klipper.bin robin_nano.bin`
it created new bin file in klipper OUT folder called Robin_Nano.bin
i renamed this newly created bin file to Robin_Nano35.bin and it worked!!!

closed 2021-01-27 by andriusmot