Klipper3d/klipper #3851: G28 Initially moves towards the bed

totalitarian, 2021-01-27:
Hi,

When I run G28 the nozzle tries to move towards the bed in Z, then home X&Y, then probes.

How can I stop it from initially moving towards the bed before homing?

```

[bltouch]
sensor_pin: ^PC2
control_pin: PA1
pin_up_touch_mode_reports_triggered: False
x_offset: -31.6
y_offset: -6.5
z_offset: 3.2
speed: 3.0
samples: 1
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False

[bed_mesh]
speed: 100
horizontal_move_z: 8
mesh_min: 50, 50
mesh_max: 300,300
probe_count: 3,3 # 3,3 or 5,5

[gcode_macro G29]
gcode: 
  G28
  BED_MESH_CALIBRATE
  BED_MESH_PROFILE SAVE=p1
  G1 X0 Y0 Z5 F4000
  
[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 350
position_max: 350
homing_speed: 100

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 350
position_max: 350
homing_speed: 100

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 4
endstop_pin:probe: z_virtual_endstop
position_max: 400
position_min: 0
homing_speed: 10

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 7.710
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 260

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 5

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
#control: pid
#pid_Kp: 690.34
#pid_Ki: 111.47
#pid_Kd: 1068.83
min_temp: 0
max_temp: 130

[heater_fan nozzle_cooling_fan]
pin: PC7

[fan]
pin: PC6

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD7054253373719862357-if00

[safe_z_home]
home_xy_position: 180, 180
speed: 100
z_hop: 10 # Move up 10mm, so the probe doesnt hit anything 
z_hop_speed: 5

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 2000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output usb_pullup_enable]
pins: !PA14

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
```

mental405, 2021-01-28:
Given your config file, it should be hopping 10mm in the positive Z direction prior to homing x and y due to the `z_hop:` option specified in `[safe_z_home]'.  Is it hopping the wrong way?

If you commend out the '[safe_z_home]' section and attempt to home _only_ x and y, does it still move the z axis?

totalitarian, 2021-01-28:
My fault, had the dir_pin set wrong

closed 2021-01-28 by totalitarian