Klipper3d/klipper #3852: Multiple hotends (with different thermistors) sharing same mcu pins

locomarco, 2021-01-27:
I am using two hotends with different thermistors.
Right now i have to edit printer.cfg everytime i switch the hotend.

To make it easier, i tried to configure extruder1, where i copied all parameters and only changed thermistor type.
Unfortunately Klipper does not allow that, because pin assignments are the same in both extruder configurations.

Is there a possibility to quickly change thermistor type, without editing the configuration file?

mental405, 2021-01-28:
If I am understanding you correctly. This is not a dual extruder printer, but a printer where you change the hotends out frequently enough to where it annoys you to update Printer.cfg when you change them out. You cannot just create a second tool because that tool uses the same pins as the first tool. So you need a terminal/gcode command like
`SET_TEMPERATURE_SENSOR heater=extruder sensor_type=BlahBlahThermistorModel`
that can be executed while the printer is running to set the thermistor.

Admittedly I haven't dug into the issue deep enough to really give an authoritative answer on it, but it does not seem like it would be too terrible to implement just swapping out the thermistor type. You would also need something to change the PID values at runtime as well.

This is something a developer would need to pick up and implement as there currently exists no support for it.

Since detaching the hotend and thermistor would cause the ADC to freak out and MCU to shutdown, a solution for the short term might be to include a separate config for your extruder and have a shell script that swaps that config out via a macro button in octoprint.

At the top of printer.cfg you would have something like
`include /extruders/extruder.cfg'

and then in that extruders folder you would have 

`v6.cfg` and `mosquito.cfg` or whatever
your shell script would so something like `cp v6.cfg extruder.cfg`
so that the proper values are loaded.

This is kind of hacky, but until someone picks up the issue and writes code for it, other than manually editing the config file I see no other way around it.

jakep82, 2021-01-28:
Not sure I see the value in implementing anything to solve this issue.  Physically changing hotends will always require a firmware_restart, and using a couple [include] lines makes the config changes trivial.

closed 2021-02-03 by klipper-gitissuebot