Klipper3d/klipper #3856: Sudden underextrusion with calibrated printer

Larvae90, 2021-01-28:
Hello,

I have printed several parts with my printer without any issues. But now I suddenly get some severe underextrusion. I didn't change the filament or any configuration or hardware. The only thing that changed recently is the ptfe tube because I had a problem with the fitting and had to replace it.

I checked that the extruder is working fine, no slipping of the drive gear on the filament, drive gear didn't come loose and no signs of step losses. I also tried an old gcode file that I know works, in case it was a slicer issue, but I get the same results. Since both of these seem to work fine I was wondering if it's a klipper issue, hence this post.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5889048/klippy.log)

mental405, 2021-01-28:
Did you make any changes to Klipper or update it before the under extrusion started?

Larvae90, 2021-01-28:
Not that I know of. Both hardware and software should be the same. I only updated klipper after the problem started, just in case. And of course I restarted both klipper and the raspberry pi several times by now.

jakep82, 2021-01-28:
Have you inspected your extruder and verified your slicer settings?  If you didn't update Klipper, then Klipper isn't the problem.

th33xitus, 2021-01-28:
Please read: https://github.com/KevinOConnor/klipper/blob/work-contact-20210111/docs/Contact.md

Since it seems that Firmware isn't the issue, this is the wrong place to ask for support.
Feel free to join the Klipper Discord.

DeZepTup, 2021-02-19:
Have you calibrated Pressure Advance?

KevinOConnor, 2021-03-05:
Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ) I am closing this issue.

-Kevin

closed 2021-03-05 by KevinOConnor