Klipper3d/klipper #3858: Must home axis first on extrusion move

ChipCE, 2021-01-29:
Sorry if I miss something!
I have some weird problem with load/unload filament script from LCD menu. The fist time I select the "Load filament", I got the <code>Must home axis first...</code> warning, but if I select that option one more time, the filament load just fine. 

This is my custom script of load/unload and menu config
<pre>

# load filament. Slow feed filament into the gear, fast load to cold zone then slow load to nozzle.
[gcode_macro M701]
default_parameter_FEED_LENGTH: 10
default_parameter_FAST_LOAD_LENGTH: 45
default_parameter_SLOW_LOAD_LENGTH: 35
gcode:
    SAVE_GCODE_STATE NAME=loading_filament
    PLAY_SINGLE_FEEDBACK_SOUND
    M117 Loading Fil.
    M83
    G92 E0.0
    MIN_TEMP_CHECK
    G1 E{FEED_LENGTH} F200    ; slow feed filament
    G1 E{FAST_LOAD_LENGTH} F2000    ; Fast load to cold zone
    G1 E{SLOW_LOAD_LENGTH} F100     ; Slow load to nozzle
    G92 E0.0
    M117 Ready
    RESTORE_GCODE_STATE NAME=loading_filament

# Unload filament. Extrude a small amount, quick pull then slow pull
[gcode_macro M702]
default_parameter_FEED_LENGTH: 10
default_parameter_FAST_UNLOAD_LENGTH: 60
default_parameter_SLOW_UNLOAD_LENGTH: 20
gcode:
    SAVE_GCODE_STATE NAME=unloading_filament
    PLAY_SINGLE_FEEDBACK_SOUND
    M117 Unloading Fil.
    MIN_TEMP_CHECK
    G91 ; set relative
    G1 E{FEED_LENGTH} F100 
    G92 E0.0
    G1 E-{FAST_UNLOAD_LENGTH} F2000  ; fast unload
    G92 E0.0
    G1 E-{SLOW_UNLOAD_LENGTH} F1000  ; slow unload
    G92 E0.0
    M117 Ready
    RESTORE_GCODE_STATE NAME=unloading_filament

[gcode_macro MIN_TEMP_CHECK T]
default_parameter_T: 200
gcode: 
    {% if printer.extruder.target != 0 %} # if there is a setpoint for extruder
        {% if printer.extruder.temperature < printer.extruder.target %} # if not reached, heat
            PLAY_LONG_FEEDBACK_SOUND
            M109 S{printer.extruder.target|float}
        {% endif %}
    {% else %} # if no setpoint for extruder
        PLAY_LONG_FEEDBACK_SOUND
        M109 S{T}
    {% endif %}
    PLAY_DOUBLE_FEEDBACK_SOUND

[menu __menutree __filament __loadf]
type: command
name: Load Fil.
gcode:
    M701

[menu __menutree __filament __unloadf]
type: command
name: Unload Fil.
gcode:
    M702

</pre>

And the M300 macro for FEEDBACK_SOUND, but I don't think it have anything to do with the problem
<pre>

[gcode_macro M300]
default_parameter_S: 100
#   Use a default 1kHz tone if S is omitted.
default_parameter_P: 1000
#   Use a 10ms duration is P is omitted.
gcode:
    SET_PIN PIN=BEEPER_pin VALUE={S}
    G4 P{P}
    SET_PIN PIN=BEEPER_pin VALUE=0

[gcode_macro PLAY_SINGLE_FEEDBACK_SOUND]
gcode:
    M300 S100 P100
    M300 S1 P100

[gcode_macro PLAY_DOUBLE_FEEDBACK_SOUND]
gcode:
    M300 S100 P100
    M300 S1 P100
    M300 S100 P100
    M300 S1 P100

[gcode_macro PLAY_LONG_FEEDBACK_SOUND]
gcode:
    M300 S100 P1000
    M300 S1 P100

</pre>

[klippy.log](https://github.com/KevinOConnor/klipper/files/5894506/klippy.log)

The--Captain, 2021-02-12:
I'm having a similar problem.  Hope my ticket isn't a duplicate, but it might be.

ChipCE, 2021-03-06:
Seem like does not happen in the newer version. I will close this issue.

closed 2021-03-06 by ChipCE