Klipper3d/klipper #3859: Dual X carriage

sedlacekdavid, 2021-01-29:
Hi Guys, anybody here who's running a dual carriage setup? 

I'm almost finished with my build -- I'm rebuilding my machine to support two X carriages and of course I didn't do any research how much this is usable. An SKR pro board & six TMC2226 drivers & two direct drive extruders.

Thanks, David.

Hywelmartin, 2021-01-30:
https://github.com/KevinOConnor/klipper/blob/work-contact-20210111/docs/Contact.md

Doccos, 2021-02-17:
Yes, I use one with success (after a brutal crash due to the klipper3d safety features)

sedlacekdavid, 2021-02-17:
Thats an excellent message! I’m almost done with my rebuild. I’ll be testing it this week.
[image]

sedlacekdavid, 2021-02-22:
Hi guys. I can confirm that everything works 100 % perfect. The setup is: Fluidd (no more octoprint after ~1 year), SKR PRO 1.2, 6x TMC2226, 6 steppers, 5 fans, 0.4 & 0.8 nozzles. There is some calibration to be done, but overall after setting all the macros I have a very good feeling about that.

So far what I tested was a print at ~ 120 mm/s without any glitches. I'm still waiting for an accelerometer.

Later on I'll post some video.

sedlacekdavid, 2021-02-22:
[image]

closed 2021-03-03 by sedlacekdavid